FarmBot/farmbot_os #50: Decrease wait time for commands

roryaronson, 2015-02-24:
Hey @TimEvWw, Rick and I were messing with the latest bot and found that commands had a delay of about a minute. We looked into the controller source code and found this:

https://github.com/FarmBot/farmbot-raspberry-pi-controller/blob/bbc816cb80ad84150404a26872ff75660da93c42/lib/controller.rb#L128

Is there a reason the `+ 60` is there?

TimEvWw, 2015-02-24:
The idea there is that the system waits for a refresh signal during one
minute (it's named 'refresh' because I use this method a lot to trigger
screen refreshes when new data arrives). If after one minute there is no
signal received, it will check the database to see if a command has been
received without setting the refresh flag. So that signal goes wrong. That
would explain it.

RickCarlino, 2015-02-24:
Do you think there will be any repercussions for lowering the number lower, like to `1`?

TimEvWw, 2015-02-24:
Higher system load mainly. SQLite was a bit sluggish for continuous polling
for my taste (hence the refresh system) but it works as a quick fix.

roryaronson, 2015-02-24:
Ok, so menu.rb is scheduling a command, but not setting the refresh flag?

I removed the +60 altogether locally and watched the CPU go to 85% haha

TimEvWw, 2015-02-25:
I think that the setting of the flag was done in the same routine as writing of the command but it was lost in refactoring. Will check.

TimEvWw, 2015-02-26:
Refresh is modified.

closed 2015-02-26 by TimEvWw