FarmBot/farmbot_os #51: @dbaccess is nil in MessageHandler

RickCarlino, 2015-02-25:
Hey Tim,

any idea why @dbaccess is nil in `MessageHandler`?

```ruby
class MessageHandler
  def initialize
    # . . . 
    @dbaccess = $dbaccess
  end
end
```

TimEvWw, 2015-02-25:
I used one shared db access stored in $dbaccess for all modules. This is normally instantiated in farmbot.rb

TimEvWw, 2015-02-25:
It just takes the RPi a long time to start up db access (seconds while on a PC this is fractions of a second) so doing it once instead of multiple times (for each class separate db access) made the software start up a bit faster.

roryaronson, 2015-02-25:
Perhaps when we upgrade to Raspberry Pi 2's performance issues like that will be less of an issue ;)

@TimEvWw, hop onto g-chat if you can!

TimEvWw, 2015-02-25:
Given that the RPi2 also runs on a SD card, and is only a few times faster, upgrading will help but won't be a cure all.

G-chat in an hour and half.

TimEvWw, 2015-02-26:
If you run it now with farmtalk.rb, dbaccess is set.

closed 2015-03-17 by RickCarlino