FarmBot/farmbot_os #157: UNABLE TO DOWNLOAD DEVICE CREDENTIALS

ralphwu, 2016-05-17:
I start the farmbot-raspberry-pi-controller as follows and the my.farmbot.io device page show UNABLE TO DOWNLOAD DEVICE CREDENTIALS

There is not credentials.yml file but it seems replaced by db/settings.yml

fa@NanoPi2:~/farmbot-raspberry-pi-controller$ ruby farmbot.rb
Connecting...
Conencting to mqtt.farmbot.io:1883
Connected!
OH NOES! Farmbot was disconnected from MQTT. Retrying...
Connecting...
Conencting to mqtt.farmbot.io:1883
Connected!

I am not sure it is connected!

RickCarlino, 2016-05-17:
@ralphwu Did you register for an account and run `ruby setup.rb`?

ralphwu, 2016-05-17:
Yes!

RickCarlino, 2016-05-17:
@ralphwu I think I need to deploy to production. One moment. I will deploy the latest version.

ralphwu, 2016-05-17:
Ok, Thks

RickCarlino, 2016-05-17:
@ralphwu Can you please try

1. Run `ruby setup.rb` again (this is a known bug- old credentials will not be valid any more, sorry)
2. Visit the [login page](http://my.farmbot.io/app/login)
3. Do a hard refresh (ctrl+shift+R)
4. Try logging in again.

Please let me know if this helps.

ralphwu, 2016-05-17:
Works perfectly! Thks!

closed 2016-05-17 by RickCarlino