FarmBot/farmbot_os #162: Unable to run farmbot.rb

dbuckingham, 2016-07-31:
I followed the instructions for installing the software on my Raspberry Pi 3, but when I try to run 'ruby farmbot.rb', I get the following error:

> pi@farmbot:~/farmbot-raspberry-pi-controller$ ruby farmbot.rb
> /home/pi/.rvm/rubies/ruby-2.3.0/lib/ruby/2.3.0/rubygems/core_ext/kernel_require.rb:120:in `require': cannot load such file -- pry (LoadError)
>         from /home/pi/.rvm/rubies/ruby-2.3.0/lib/ruby/2.3.0/rubygems/core_ext/kernel_require.rb:120:in `require'
>         from /home/pi/.rvm/gems/ruby-2.3.0/gems/activesupport-4.2.3/lib/active_support/dependencies.rb:274:in `block in require'
>         from /home/pi/.rvm/gems/ruby-2.3.0/gems/activesupport-4.2.3/lib/active_support/dependencies.rb:240:in `load_dependency'
>         from /home/pi/.rvm/gems/ruby-2.3.0/gems/activesupport-4.2.3/lib/active_support/dependencies.rb:274:in `require'
>         from /home/pi/farmbot-raspberry-pi-controller/lib/telemetry_message.rb:1:in `<top (required)>'
>         from /home/pi/farmbot-raspberry-pi-controller/lib/command_objects/send_message.rb:3:in `require_relative'
>         from /home/pi/farmbot-raspberry-pi-controller/lib/command_objects/send_message.rb:3:in `<top (required)>'
>         from /home/pi/farmbot-raspberry-pi-controller/lib/command_objects/commands.rb:17:in `require_relative'
>         from /home/pi/farmbot-raspberry-pi-controller/lib/command_objects/commands.rb:17:in `<top (required)>'
>         from lib/controllers/exec_sequence_controller.rb:2:in `require_relative'
>         from lib/controllers/exec_sequence_controller.rb:2:in `<top (required)>'
>         from /home/pi/farmbot-raspberry-pi-controller/lib/command_objects/resolve_controller.rb:1:in `load'
>         from /home/pi/farmbot-raspberry-pi-controller/lib/command_objects/resolve_controller.rb:1:in `block in <top (required)>'
>         from /home/pi/farmbot-raspberry-pi-controller/lib/command_objects/resolve_controller.rb:1:in `each'
>         from /home/pi/farmbot-raspberry-pi-controller/lib/command_objects/resolve_controller.rb:1:in `<top (required)>'
>         from /home/pi/farmbot-raspberry-pi-controller/lib/messaging/message_handler.rb:6:in `require_relative'
>         from /home/pi/farmbot-raspberry-pi-controller/lib/messaging/message_handler.rb:6:in `<top (required)>'
>         from /home/pi/farmbot-raspberry-pi-controller/lib/farmbot-pi.rb:3:in `require_relative'
>         from /home/pi/farmbot-raspberry-pi-controller/lib/farmbot-pi.rb:3:in `<top (required)>'
>         from farmbot.rb:1:in `require_relative'
>         from farmbot.rb:1:in `<main>'
> pi@farmbot:~/farmbot-raspberry-pi-controller$
> 

Any one understand why farmbot.rb cannot run?

RickCarlino, 2016-07-31:
# The Problem

You caught me, I'm guilty. This is a real bug. I use [Pry](http://pryrepl.org/) for live debugging in dev mode (on an x86). Unfortunately, Pry does not install on all ARM based CPUs, so I don't put it in the gemfile. I probably forgot to take out a `require "pry"` statement somewhere.

## Quick Fix

 1. `gem install pry`
 2. Start over with install instructions

## Real Fix

 1. Remove all references to `pry` in code base and be more careful in the future, Rick.

I will do that here in a moment. Thanks for pointing it out. As a side note, if you're trying to run this code on a _real_ bot, consider using one of the stable releases under the "releases" tab. We don't guarantee that latest master is stable.

Thanks for the heads up! Feel free to drop an issue on anything else you find.

closed 2016-07-31 by RickCarlino

dbuckingham, 2016-08-01:
Thanks for the reply!  I was able to find the reference to pry (lib/telemetry_message.rb), and after commenting out the offending line, the software appears to run.  Actually, it fails to connect to the Arduino MEGA, but that's ok since I don't have one connected. ;-)

FYI...  I followed instructions here: https://farmbot-software.readme.io/docs/raspberry-pi-software and in the repo README.md file.  It doesn't suggest checking out a release tag, or otherwise changing to any other branch.

Thanks again,

David

commit 826db2ac (2016-08-01): UPDATE README. Fixes #162 .