FarmBot/farmbot_os #165: Farmbot Setup Errors

riaandeyzel, 2016-08-08:
Good day

I have successfully setup the Arduino Controller, but I keep getting this error when trying to launch Farmbot:

```
Want to run ruby farmbot.rb now?
y
Connecting...
Conencting to mqtt.farmbot.io:1883
Connected!
/usr/local/rvm/gems/ruby-2.3.0/gems/em-mqtt-0.0.5/lib/em/mqtt/client_connection.rb:181:in connect_ack': Connection refused: not authorised (MQTT::ProtocolException)
from /usr/local/rvm/gems/ruby-2.3.0/gems/em-mqtt-0.0.5/lib/em/mqtt/client_connection.rb:163:inprocess_packet'
from /usr/local/rvm/gems/ruby-2.3.0/gems/em-mqtt-0.0.5/lib/em/mqtt/connection.rb:37:in receive_data'
from /usr/local/rvm/gems/ruby-2.3.0/gems/eventmachine-1.2.0.1/lib/eventmachine.rb:194:inrun_machine'
from /usr/local/rvm/gems/ruby-2.3.0/gems/eventmachine-1.2.0.1/lib/eventmachine.rb:194:in run'
from /root/farmbot-raspberry-pi-controller/lib/farmbot-pi.rb:29:instart'
from farmbot.rb:5:in <top (required)>'
from setup.rb:62:inload'
from setup.rb:62:in `'

```

Kind Regards,
Riaan

RickCarlino, 2016-08-09:
Hi, Riaan. Thanks for letting us know!

It's possible that you're using the latest version ("edge") instead of a [stable release](https://github.com/FarmBot/farmbot-raspberry-pi-controller/releases). If so, that's OK- just make sure you register for an account at http://staging.farmbot.io instead of http://my.farmbot.io .

Could you try [registering for an account on the staging server](http://staging.farmbot.io/app/login) and using that server instead of my.farmbot.io ? Most likely, you're using a newer version of FarmBot than the stable production server can support.

RickCarlino, 2016-08-09:
If this does not resolve your issue, please let me know and I can reopen.

closed 2016-08-09 by RickCarlino