FarmBot/farmbot_os #167: Connection refused after starting Farmbr.rb

paulch9, 2016-08-16:
Dear Farbot Team,

after i install the Farmbot raspberry pi controller (Alpha1.1) software, i tried to run it. But after a short connection to the Arduino (Alpha 1.0). The connection was refused withe the error text " not authorized". (MQTT::Protocol Exception). Have i installed the wrong version of PI controller, or is something wrong withe the UUID? 

Best regards 

Christoph

RickCarlino, 2016-08-16:
@paulch9 Am I correct in assuming you already registered for a username/password at `my.farmbot.io`?

Could you try registering for an account at `staging.farmbot.io` and then running setup.rb again (this time, provide `staging.farmbot.io` as the server).

Please let me know if that helps or if you have any other information to share. I will re-open if those steps don't help- please just let me know.

closed 2016-08-16 by RickCarlino

paulch9, 2016-08-23:
Hello Rick, the staging.farmbot.io app works very well. Thanks a lot. But one little Question i have already. If i try to move the Farmbot with the arrow buttons on the app, nothing happens and the message "no input handler method exist" do you have any idea's whats wrong? 

Best regards 

Christoph

RickCarlino, 2016-08-23:
@paulch9 Is the device actually moving though? Sometimes the firmware tries to send messages to the RPi that are not yet implemented. Even though no input handler exists for an incoming GCode, it might not have any adverse effects on actual software operation. It's mostly just good to know that a message is coming in rather than ignoring it.

I can re-word that to sound a little less scary. Do you think you could paste in the exact text of the message you got just to confirm that if that's the issue?

paulch9, 2016-08-23:
Hello Rick, no the device isn't moving. The exact text after i push the arrow button.
"reporting_end_stops is a valid GCode, but no input handler exists". If i touch the stepper i can feel that it wants to run, but it doesn't move.

Best regards

Christoph

RickCarlino, 2016-08-23:
OK So there are probably two things going on:

Firs thing: The bot is sending the pi a `reporting_end_stops` message, which the RPI is basically just ignoring. Nothing to worry about there.

Second thing: If the motors are moving (but not properly) it may be an adjustment issue. I hit this one myself a while back. Have you tried adjusting those small screws on the RAMPS board yet? Sometimes they need some tweaking before they will actually move a motor. Otherwise, the motors will just flinch and the drivers will get very hot.

paulch9, 2016-08-23:
Hello Rick, it works.. one Axis is now moving. The issue with the screw on the RAMPS was the problem. But by the way, how can i define in the Web App the point "STEPS PER MM" ? There is a different between the length witch i would drive and the real driven way length. 
Sorry to be so irritating

Best regards 

Christoph

roryaronson, 2016-08-23:
Hi Christoph,

Glad you figured out the first issue. The STEPS PER MM value is not currently configurable from the web app, and that is definitely a problem we need to solve soon. Right now my bot moves only 1/5 of the distance it is supposed to because of the motor mechanics, belt pitch, and pulley teeth. We'll hopefully get a fix for this out in the next few weeks.