FarmBot/farmbot_os #168: bundle install failing to install json (1.8.3)

dwhickox, 2016-08-29:
just running bundle install
```
pi@raspberrypi:~/farmbot-raspberry-pi-controller $ sudo bundle install
Don't run Bundler as root. Bundler can ask for sudo if it is needed, and
installing your bundle as root will break this application for all non-root
users on this machine.
Fetching gem metadata from https://rubygems.org/
Fetching version metadata from https://rubygems.org/
Fetching dependency metadata from https://rubygems.org/
Installing rake 10.4.2
Installing i18n 0.7.0
Installing json 1.8.3 with native extensions

Gem::Ext::BuildError: ERROR: Failed to build gem native extension.

    /usr/bin/ruby2.1 extconf.rb
mkmf.rb can't find header files for ruby at /usr/lib/ruby/include/ruby.h

extconf failed, exit code 1

Gem files will remain installed in /var/lib/gems/2.1.0/gems/json-1.8.3 for inspection.
Results logged to /var/lib/gems/2.1.0/extensions/arm-linux/2.1.0/json-1.8.3/gem_make.out
Installing minitest 5.8.0
Installing thread_safe 0.3.5
Installing builder 3.2.2
Installing erubis 2.7.0
Installing mini_portile 0.6.2
Installing rack 1.6.4
Installing arel 6.0.3
Installing thor 0.19.1
Installing diff-lcs 1.2.5
Installing docile 1.1.5
Installing unf_ext 0.0.7.2 with native extensions

Gem::Ext::BuildError: ERROR: Failed to build gem native extension.

    /usr/bin/ruby2.1 extconf.rb
mkmf.rb can't find header files for ruby at /usr/lib/ruby/include/ruby.h

extconf failed, exit code 1

Gem files will remain installed in /var/lib/gems/2.1.0/gems/unf_ext-0.0.7.2 for inspection.
Results logged to /var/lib/gems/2.1.0/extensions/arm-linux/2.1.0/unf_ext-0.0.7.2/gem_make.out
Installing eventmachine 1.2.0.1 with native extensions

Gem::Ext::BuildError: ERROR: Failed to build gem native extension.

    /usr/bin/ruby2.1 extconf.rb
mkmf.rb can't find header files for ruby at /usr/lib/ruby/include/ruby.h

extconf failed, exit code 1

Gem files will remain installed in /var/lib/gems/2.1.0/gems/eventmachine-1.2.0.1 for inspection.
Results logged to /var/lib/gems/2.1.0/extensions/arm-linux/2.1.0/eventmachine-1.2.0.1/gem_make.out
Installing mqtt 0.3.1
Installing mime-types 2.99.1
Installing netrc 0.11.0
Installing serialport 1.3.1 with native extensions

Gem::Ext::BuildError: ERROR: Failed to build gem native extension.

    /usr/bin/ruby2.1 extconf.rb
mkmf.rb can't find header files for ruby at /usr/lib/ruby/include/ruby.h

extconf failed, exit code 1

Gem files will remain installed in /var/lib/gems/2.1.0/gems/serialport-1.3.1 for inspection.
Results logged to /var/lib/gems/2.1.0/extensions/arm-linux/2.1.0/serialport-1.3.1/gem_make.out
Installing god 0.13.6 with native extensions

[... 409 lines cut ...]
extconf failed, exit code 1

Gem files will remain installed in /tmp/bundler20160829-20303-17c65wpgod-0.13.6/gems/god-0.13.6 for inspection.
Results logged to /tmp/bundler20160829-20303-17c65wpgod-0.13.6/extensions/arm-linux/2.1.0/god-0.13.6/gem_make.out
Using highline 1.7.8
Using ice_cube 0.13.0
Using liquid 3.0.6
Using rspec-support 3.3.0
Using settingslogic 2.0.9
Using simplecov-html 0.10.0
Installing sqlite3 1.3.10 with native extensions

Gem::Ext::BuildError: ERROR: Failed to build gem native extension.

    /usr/bin/ruby2.1 extconf.rb
mkmf.rb can't find header files for ruby at /usr/lib/ruby/include/ruby.h

extconf failed, exit code 1

Gem files will remain installed in /tmp/bundler20160829-20303-opn966sqlite3-1.3.10/gems/sqlite3-1.3.10 for inspection.
Results logged to /tmp/bundler20160829-20303-opn966sqlite3-1.3.10/extensions/arm-linux/2.1.0/sqlite3-1.3.10/gem_make.out
Using bundler 1.12.5
An error occurred while installing json (1.8.3), and Bundler cannot continue.
Make sure that `gem install json -v '1.8.3'` succeeds before bundling.
```

gabrielburnworth, 2016-08-29:
Did you follow all the steps in the readme under [Installation on a Raspberry Pi](https://github.com/FarmBot/farmbot-raspberry-pi-controller/blob/master/README.md#installation-on-raspberry-pi)?

I don't see `sudo apt-get remove ruby* --purge` in the output you pasted. Also, do not use `sudo` for `gem install bundler` or `bundle install`. The errors are likely due to the wrong ruby version (it seems to by trying to use ruby 2.1 `/usr/bin/ruby2.1` while at least version 2.2 is needed) and bundle install having been run as root (`Permission denied`). You may need to start fresh and follow the instructions as linked above.

RickCarlino, 2016-08-29:
Side note for anyone who found this issue via searching: This one may also be due to the fact that he installed RVM using LXTerm on the GUI side of the Pi.

He is currently [working through these steps](https://rvm.io/integration/lxterminal) to see if it helps calm the RVM issues.

gabrielburnworth, 2016-08-29:
Yes, if installing via LXTerminal in the GUI and you get `bash: gem: command not found`, you'll need to run `lxterminal -e "bash -il"` and then continue installation in the new terminal that opens:
```
cd farmbot-raspberry-pi-controller
gem install bundler
bundle install
rake db:setup
ruby setup.rb
```

closed 2016-10-04 by ConnorRigby