FarmBot/farmbot_os #169: MQTT error

blaimentin, 2016-09-07:
hello, i have installed your firmware alpha 1.1 for my raspberry 3 and arduino with the last version,
in the instal everything is ok i run ruby setup.rb but in http://my.farmbot.io/ there an error impossible to conect to mqtt server can you help me?

I started to build my farmbot is a very great project thank you for that 
sorry for my english 
clement to france

RickCarlino, 2016-09-08:
Hi @blaimentin . Please try using `staging.farmbot.io` instead. The problem will be fixed in the next 2 weeks. *Also make sure to run `setup.rb` again pointing to that server.*

closed 2016-09-08 by RickCarlino

blaimentin, 2016-09-13:
hello, its working in staging farmbot.io thank you

RickCarlino, 2016-09-13:
Glad to hear that @blaimentin . I am upgrading the staging servers today so you may experience some outages for the next 6 hours. Everything should be back to normal tomorrow.