FarmBot/farmbot_os #171: error connect server

airlinepilot, 2016-09-24:
I do not connect Farmot Raspicontroler to my server. I get an error like this.
Setup is complete! You may now runruby farmbot.rbto start the bot. Want to runruby farmbot.rb` now?
y

Connecting...
/usr/local/rvm/gems/ruby-2.3.1/gems/farmbot-resource-0.6.0/lib/resources/abstract_resource.rb:31:in fetch_no': undefined methodurl' for #Errno::ECONNREFUSED:0xebcef0 (NoMethodError)
from /usr/local/rvm/gems/ruby-2.3.1/gems/farmbot-resource-0.6.0/lib/resources/abstract_resource.rb:23:in block in fetch' from /usr/local/rvm/gems/ruby-2.3.1/gems/farmbot-resource-0.6.0/lib/http/bad_http_resp.rb:3:inno'
from /usr/local/rvm/gems/ruby-2.3.1/gems/farmbot-resource-0.6.0/lib/resources/abstract_resource.rb:23:in fetch' from /usr/local/rvm/gems/ruby-2.3.1/gems/farmbot-resource-0.6.0/lib/resources/abstract_resource.rb:41:inall'
from /usr/local/rvm/gems/ruby-2.3.1/gems/farmbot-resource-0.6.0/lib/resources/device.rb:9:in current' from /var/www/farmbot-raspberry-pi-controller/lib/farmbot-pi.rb:21:ininitialize'
from farmbot.rb:5:in new' from farmbot.rb:5:in'
from setup.rb:62:in load' from setup.rb:62:in
'

RickCarlino, 2016-09-24:
A few questions:

 1. Which server are you trying to connect to?
 2. Did you sign up for an account on that server using the browser already?
 3. Did you use one of the [stable release versions](https://github.com/FarmBot/farmbot-raspberry-pi-controller/releases)?

airlinepilot, 2016-09-26:
1. I get an error when i connect my server but your server not error.
2. Yes I do.
3. yes I do.

airlinepilot, 2016-09-26:
In file farmbot-raspberry-pi-controller/lib/farmbot-pi.rb
` def initialize
#    @rest_client    = FBPi::RPiRestClient.new(WEBAPP_URL)
    @rest_client    = FBPi::RPiRestClient.new("http://jungle.com.vn:5000") // I set for my server 
    uuid            = @rest_client.device.current["uuid"]  // this is error 
    token           = @rest_client.config.token
    @mqtt           = MQTTAdapter.new(uuid, token)
    @status_storage = FBPi::StatusStorage.new("db/bot_status.pstore")
    @bot = FBPi::BotDecorator.build(@status_storage, @mqtt, @rest_client)
  end
`
Do you think like me ?

RickCarlino, 2016-09-26:
@airlinepilot The RPi controller was built to run on other servers- you don't need to change the source code to use your own server. We use different servers when we develop FarmBot. For example, I have a local server, a staging server and a public server. When I need to use the public server instead of my local server, I just re-run `setup.rb`.

Did you run `setup.rb` already? If so, did you include the port number? Make sure you type your API URL in this manner: `your-server.com:3333` (exclude protocol, but do include port number).

Thanks for bringing the issue to our attention- please let me know if that helps.

closed 2016-10-04 by ConnorRigby