FarmBot/farmbot_os #186: Unable to save Wifi configuration

mrmadhu, 2016-10-15:
Hello 

I'm not sure if this was discussed already, as per the setup instruction when i try to configure my home wifi  details and save by accessing 192.168.24.1, nothing happens after clicking save configuration button, it stays in the initial screen, i tried using the both the .img file 2.0.3 and 2.0.2  both has the same issue.  Thanks in advance for any help

mrmadhu, 2016-10-15:
Only SSID's that are visible is as following 

test_ssid
meep
lawl

it does not show my home wifi SSID

ConnorRigby, 2016-10-15:
ah yeah sorry. those things showing up should never happen on a real Raspberry pi. Im going to push a new release you can try fairly shortly.

jokosup, 2016-10-18:
any ETA when can we expect the new release?

ConnorRigby, 2016-10-19:
Your on a pi3 correct?
this release should work: [here](https://github.com/FarmBot/farmbot_os/releases/download/v2.0.5/fw-v2.0.5.img)

mrmadhu, 2016-10-19:
Sorry, It dosent work, it does not scan and show the home wifi SSID

RickCarlino, 2016-10-19:
@mrmadhu Can you try opening the [JavaScript console](https://developers.google.com/web/tools/chrome-devtools/console/) and tell me if you see any errors? That might have additional information about what the issue is.

RickCarlino, 2016-10-19:
Javascript console is `ctrl + shift + J` on Linux and Windows. I forget what the hotkey is on Mac.

mrmadhu, 2016-10-19:
Please find the attached screen shot 
[image]

jokosup, 2016-10-20:
Thanks Rick,

1. I am using ubuntu and windows 7.
2. Ubee cable modem with router
3. no firewall software.

i did try with setting up my firewall protection to low in my router as well.

jokosup, 2016-10-20:
The wifi sometime work, sometime doesn't.

[image]

ConnorRigby, 2016-10-20:
Ah i see. That is actually a common problem so far. It is a power issue. The PI3 requires more power than previous pis. Something to the tune of 2.5 amps

mrmadhu, 2016-10-20:
hello 
i'm still stuck, with the same problem even after disabling firewall and higher amps power supply 
i'm using a raspberry pi 3 
is there any way to ssh to the farmbot os, if yes which ip address ?
Madhu

   
Closed #186.—

ConnorRigby, 2016-10-20:
There is no ssh into it yet. 
(tho it is a feature we are working on)

But i made it on the latest release be able to plug a mouse and keyboard and use the shell over hdmi. 

What is the amperage of you power supply?

jokosup, 2016-10-21:
i am using 2.1A usb from the power plug. 

it works fine when i run Raspbian Jessie, but not the farmbot_os. 
what makes farmbot_os needs more power?

ConnorRigby, 2016-10-21:
Im beginning to think it is not a power issue. 2.1 amps should be enough. I can't reproduce locally so i am going to be testing some things on other peoples environments.

ConnorRigby, 2016-10-21:
relevant kernel logs: 
```
brcmfmac: brcmf_sdio_drivestrengthinit: No SDIO Drive strength init done for chip 43430 rev 1 pmurev 24
brcmfmac_sdio mmc0:0001:1: Direct firmware load for brcm/brcmfmac4330-sdio.txt failed with error -2
brcmfmac: brcmf_sdio_htclk: HT Avail timeout (1000000): clkctl 0x50
```

RickCarlino, 2016-10-21:
@jokosup @mrmadhu We're still investigating, but thanks for all the great information so far. It appears to be a driver issue.

ConnorRigby, 2016-10-21:
@jokosup can you try to boot your pi without a hdmi cable plugged in?

jokosup, 2016-10-21:
yes, i have tried that. same results.

mrmadhu, 2016-10-21:
Same result  for me as well

ConnorRigby, 2016-10-21:
@jokosup and @mrmadhu 

can you guys please try flashing the pre-release for me? 
[get it here](https://github.com/FarmBot/farmbot_os/releases/tag/v2.0.7-alpha)
Try it with and without HDMI plugged in.

mrmadhu, 2016-10-21:
Hi 
just tried, its able to connect to home wifi and gets an IP 

but no luck with my.farmbot.io connectivity or no way to login to the OS, via SSH 

Madhu

ConnorRigby, 2016-10-21:
@mrmadhu have you created an account on my.farmbot.io first?

mrmadhu, 2016-10-21:
yes and the id was configured as on the wificonfigurator page

ConnorRigby, 2016-10-21:
can you try creating an account on [http://staging.farmbot.io](http://staging.farmbot.io) and using those same credentials on configurator?

mrmadhu, 2016-10-21:
do i need to re-image the sd card again ?

ConnorRigby, 2016-10-21:
If wifi configurator comes up you dont, but if its 'bricked' then you will

mrmadhu, 2016-10-21:
@ConnorRigby it connects to staging and gets a message that "BOT status OK , but no control of stepper

jokosup, 2016-10-21:
I am able to connect to the FarmbotConfigurator, but when i click Save Configuration, it got the "TRYING TO LOGIN" response.  then lost the WiFi. 
i deleted the files from AppData and restart it again, WiFi came back.

mrmadhu, 2016-10-22:
@ConnorRigby , still stuck, i'm able to shutdown pi from the web,  and there is a success message for every command send from the web like absolute x move, but no movement at the stepper, any way to find out what exactly is happening,

jokosup, 2016-10-22:
i am having the same issue as @mrmadhu. no movement at the stepper.

ConnorRigby, 2016-10-23:
sorry guys my last comment got lost in transit I guess. 

Two things. 
1. Your Arduins need an update. The hardware is all working fine, the Arduino just doesn't have matching firmware. If you guys run Linux I can walk you thru updating them being that In the alpha release I gave you, the OTA updates are broken. 

2. The wifi issue is in fact power related. If it happens again you will need to source a good 5v 2.5 amp power supply.

mrmadhu, 2016-10-23:
@ConnorRigby will get a linux Vbox, request you to please share the steps

ConnorRigby, 2016-10-23:
ok. 
* install a ubuntu 16.04 image in vbox
* Make sure you can access your arduino from `inside` the vbox. 
* find the location of your arduino. (from a terminal you can run ` ls /dev/ttyACM0` or `ls /dev/ttyUSB0`)
   * it will probably be one of:
       * /dev/ttyACM0 
       * /dev/ttyUSB0
* run this inside a terminal: ` wget https://gist.githubusercontent.com/ConnorRigby/b8acfeb91d1c087680efeb489d9c926c/raw/c89db6f40e44fca484635007bf7b1733ecf7fd64/update.sh -qO- | bash -s `

ConnorRigby, 2016-10-23:
@mrmadhu @jokosup 
I made a [beta release](https://github.com/FarmBot/farmbot_os/releases/tag/v2.0.7-beta) you can try instetad of the above steps.
 When your bot boots you should be able to hit that button and it will update your arduino. 
[image]

# NOTE
Do NOT hit the Controller Update button. It will actually make your bot go back to an older release. (because the image your using is beta)

jokosup, 2016-10-24:
unable to update the Firmware, it seems stuck in updating FW

[image]

jokosup, 2016-10-24:
by the way, the stepper motor is now working. toggle switch tools (vacuum, water valve and led) are working as well. I only tested one motor as i only have 1 stepper motor driver right now (will have more tomorrow). Camera does not work, what should i put the value for camera's url?

roryaronson, 2016-10-24:
@jokosup, the camera widget on the web app works with any webcam stream URL, but you have to setup a stream yourself. Right now the OS does not automatically stream the pi camera or a USB camera video, though that would be a nice feature! In the meantime if you set up something external, you can use that.

By the way, do you know the reason why the motors were not working for you before?

mrmadhu, 2016-10-24:
still stuck with stepper motor issue, even after updating with new firmware. @ConnorRigby  - the new image that you have uploaded with arduino fix has wifi configuration issue again

jokosup, 2016-10-24:
@roryaronson ok.  the new firmware fixed it. 

@mrmadhu try update the Arduino firmware with this command:
wget https://gist.githubusercontent.com/ConnorRigby/b8acfeb91d1c087680efeb489d9c926c/raw/c89db6f40e44fca484635007bf7b1733ecf7fd64/update.sh -qO- | bash -s
and also deploy the beta version of farmbot_os.

ConnorRigby, 2016-10-24:
So the WiFi issue is power related actually. My latest beta image just shaves some power requirements off. You need at MINIMUM 2.5 amps from a 5 volt power supply. We will be shipping with a 3 amp psu.

mrmadhu, 2016-10-25:
Perfect @ConnorRigby  all well with the new image, all works well with 2.1 amps, great job ....

mrmadhu, 2016-10-25:
there are issues with web app, not sure if we need to put this into a new issue

RickCarlino, 2016-10-25:
@mrmadhu Yes, please do. Preferably at the Repository for the frontend: https://github.com/FarmBot/farmbot-web-frontend

Thanks for all your help so far.

ConnorRigby, 2016-10-25:
Thanks for your patience guys!

closed 2016-10-25 by ConnorRigby

ConnorRigby, 2016-10-25:
@mrmadhu Ah this is good news. Your arduino needs an update though and
github is down due to DNS errors i believe. (Github hosts our arduino
firmware releases)

@joko when you connect, you will loose wifi. (The raspberry pi can not
operate in wifi hotspot mode, and client mode at the same time) Does it
scan for networks? if you provided good credentials you will no longer see
FarmbotConfigurator.

ConnorRigby, 2016-10-25:
Yes unfortunately your arduinos will need an update but if i remember
correctly the button to make them update is broken. If you guys run linux i
can walk you thru updating them from your normal pc

jokosup, 2016-10-25:
@ConnorRigby i found that my router setting needs to be adjusted. the PSK is not enabled in my guest network.  

Do you make any updates on the staging site this morning, none of the features are working. BOT is connected, but when clicking any buttons or setting, it throwing an errors.

ConnorRigby, 2016-10-25:
@jokosup  you probably need to reboot your bot.

jokosup, 2016-10-26:
i have tried to reboot several times and still not working. 

[image]