FarmBot/farmbot_os #187: Stuck on the random: nonblocking pool is initialized

jokosup, 2016-10-15:
try different SD-Cards and having same issues, work fine with NOOBS OS.

any suggestions?

ConnorRigby, 2016-10-15:
That shoudlnt be an error its just a message from the kernel. How are you seeing that message? ftdi or HDMI? 

HDMI is currently not supported without recompiling the system
if it is over FTDI can you paste the full log?

jokosup, 2016-10-16:
oh ok, i am seeing it when connecting it via HDMI. 

also, i don't see FarmbotConfigurator WiFi either.

ConnorRigby, 2016-10-16:
your on a pi3 right? can you snap a picture of it via hdmi?

jokosup, 2016-10-16:
yes, here's what i see in the screen via hdmi

[image]

ConnorRigby, 2016-10-16:
ok. So what are you powering your pi with? it looks like its not getting enough power to boot the Wifi Chip relialably

jokosup, 2016-10-16:
powered from my laptop's usb, let me try to use dedicated power.
thanks for your quick response.

ConnorRigby, 2016-10-16:
no problem. I have occasionally got this before but it goes away if i power it with something more powerful (something in the line of 2-2.5 amps) Im working on a fix so it should work at lower voltages more reliably

jokosup, 2016-10-16:
yup, that's it.. it is working now. thanks. have a great weekend.

closed 2016-10-16 by jokosup