FarmBot/farmbot_os #223: Unable to boot on Raspberry Pi 2

joevo2, 2017-01-05:
Is there a hard dependency on RPI3? I tried to boot the latest release of Farmbot OS on RPI2 but it wouldnt work I only get the rainbow boot screen

ConnorRigby, 2017-01-06:
It shouldn't be to hard to make a config for rpi2, but Farmbot's configuration relies on a wireless interface, which previous generations don't have. If you have a usb wireless dongle, i can probably set something up

joevo2, 2017-01-06:
Can we do this via Ethernet cable? :(

ConnorRigby, 2017-01-06:
yeah you would be able to, but you would have to access farmbot configuration by your farmbot's ip address. Not impossible, you will just have to find your ip address information. i will look into adding support for RPI1 and RPI2 soon

joevo2, 2017-01-06:
That wouldn't be a problem :) just need to able to boot the farmbot os on
my rpi 2. Is there anything I can help with? When is the eta for rpi2
support? Cause is kinda urgent :( sorry for the trouble

ConnorRigby, 2017-01-06:
im starting work on it right now. I don't own an RPI2 any longer, so i may need some help debugging.

joevo2, 2017-01-06:
Thanks a lot Connor appreciate it. Sure no problem about that. We will
provide assistance in debugging 😊 anything just let me know but I will
respond ASAP cause is kinda midnight here living in GMT +8

joevo2, 2017-01-08:
Any commit or fork that I can test? Sorry for being so rush. Need to run it
on rpi2 asap to
meet our milestone :(

ConnorRigby, 2017-01-08:
if you have a rather beefy linux machine you can build [this](https://github.com/ConnorRigby/farmbot_os/tree/celery_script): 
see `BUILDING.md` for instructions. Please be warned that this is probably pretty unstable at the moment.

menghong777, 2017-01-09:
Hi, @ConnorRigby . I was trying to build the RPi2 image on Ubuntu 16.10. I got an error when I tried to run "mix firmware", it returns "** (Mix) Could not locate System". Do you have any idea with this error? =)
```
using rpi2 configuration.
warning: variable "deps" does not exist and is being expanded to "deps()", please use parentheses to remove the ambiguity or change the variable name
  /home/menghong/farmbot-image/farmbot_os/deps/rpi2/nerves_lib/mix.exs:11

warning: variable "deps" does not exist and is being expanded to "deps()", please use parentheses to remove the ambiguity or change the variable name
  /home/menghong/farmbot-image/farmbot_os/deps/rpi2/nerves_lib/mix.exs:14

warning: variable "deps" does not exist and is being expanded to "deps()", please use parentheses to remove the ambiguity or change the variable name
  /home/menghong/farmbot-image/farmbot_os/deps/rpi2/amnesia/mix.exs:7

warning: variable "package" does not exist and is being expanded to "package()", please use parentheses to remove the ambiguity or change the variable name
  /home/menghong/farmbot-image/farmbot_os/deps/rpi2/amnesia/mix.exs:8

warning: variable "docs" does not exist and is being expanded to "docs()", please use parentheses to remove the ambiguity or change the variable name
  /home/menghong/farmbot-image/farmbot_os/deps/rpi2/nerves_firmware/mix.exs:17

** (Mix) Could not locate System
```

ConnorRigby, 2017-01-09:
sorry about that, the `system` needs to be built first. for clarity when we say `system` it means the root filesystem built by `buildroot` that boots linux.
so to build the system `cd` to the root of the `farmbot_os` dir, then do this: 
```bash
export MIX_ENV=prod
export NERVES_TARGET=rpi2
mix farmbot.system # this builds teh environment for building the system.
cd apps/NERVES_SYSTEM_${NERVES_TARGET}
make
```
that is going to take a while depending on your machine. once it is finished do this: 
```bash
cd ../farmbot
mix firmware
```

menghong777, 2017-01-10:
Thanks for your reply and instructions. =) Before running the `make` command, I followed Nerves Project website's instructions (https://hexdocs.pm/nerves/systems.html#creating-or-modifying-a-nerves-system-with-buildroot) then I ran `mix firmware` command. After that I got an error shown below.
```
==> nerves_network_interface
cc -c -I/usr/lib/erlang/usr/include -O2 -Wall -Wextra -Wno-unused-parameter -pedantic -std=c99 -D_XOPEN_SOURCE=600 -o src/netif.o src/netif.c
src/netif.c:26:27: fatal error: libmnl/libmnl.h: No such file or directory
 #include <libmnl/libmnl.h>
                           ^
compilation terminated.
Makefile:51: recipe for target 'src/netif.o' failed
make: *** [src/netif.o] Error 1
could not compile dependency :nerves_network_interface, "mix compile" failed. You can recompile this dependency with "mix deps.compile nerves_network_interface", update it with "mix deps.update nerves_network_interface" or clean it with "mix deps.clean nerves_network_interface"
==> farmbot
** (Mix) Could not compile with "make" (exit status: 2).
Depending on your OS, make sure to follow these instructions:

  * Mac OS X: You need to have gcc and make installed. Try running the
    commands "gcc --version" and / or "make --version". If these programs
    are not installed, you will be prompted to install them.

  * Linux: You need to have gcc and make installed. If you are using
    Ubuntu or any other Debian-based system, install the packages
    "build-essential". Also install "erlang-dev" package if not
    included in your Erlang/OTP version. If you're on Fedora, run
    "dnf group install 'Development Tools'".
```
I'm sure I have installed gcc, make, build-essential and erlang-dev. This is my first time compiling an operating system. Am I doing it right? =)

RickCarlino, 2017-01-10:
@menghong777 You will need to [install libmnl, also](https://github.com/FarmBot/farmbot_os/issues/222).

ConnorRigby, 2017-01-11:
@menghong777 @joevo2 You'll both be happy to know Raspberry Pi2, Pi1, and Pi0 support is now stable. Next release will include support for it, though you will still need to build it yourself. We are working hard to make sure our building documentation is ready for new comers also. Check back soon!

joevo2, 2017-01-11:
Wow thanks man!!! I'm super happy to hear that. Thanks a lot Connor 😊😊
wow wonder how it'll perform in RPI0

ConnorRigby, 2017-01-11:
@joevo2 i don't have one to test, but FarmbotOS itself only uses about 18 megs of ram and 10 megs of disk space. (the disk space one is subjective depending on how you look at it). 
After `linux` passes control to `Erlang` all the cpus (4 on rpi3, 1 otherwise) sit idle at about 0.93 mhz, and consume about 4.77 volts. FBOS is pretty lightweight for something considered an _OperatingSystem_

menghong777, 2017-01-11:
Thanks everyone! Really appreciate everyone efforts in this project and to helping me out! I'm building it using the latest source code now. Hope the compile process is running smooth as expected! =D

joevo2, 2017-01-11:
Awesome 👏 just wondering the image recognition for weeding is done on rpi
or on the server side? We are planning to fork or build a new algorithm for
a tray seeding function for the farmbot. Where it recognise the socket in
the trays and seed precisely in the middle of the trays socket

ConnorRigby, 2017-01-11:
@joevo2 we are actually starting work on that right now. We plan to have it happen on the rpi i believe. The current implementation happens in python right now.

joevo2, 2017-01-11:
You mean that the weeding function doesn't really work now? It only runs on
python? Not on RPI or even the web app?

roryaronson, 2017-01-11:
We have a prototype made that uses Python and OpenCV to process the images to detect all plants. That code could run anywhere (laptop, pi, in the cloud, etc) though we're planning on having it run on the pi. The code currently outputs coordinates and sizes of all plants (both weeds and our desired crops). This list will then need to be cross-listed with the list of current crops so that we can categorize what is a weed and what is our veggies. Then the user will be able to schedule sequences to run on the weeds to remove them.

Here is a mockup of what it will look like on the web app for the user to "tune" the processing algorithms. It is a two step process: first is selecting the range of colors to isolate and look at, and second is the openCV stuff that locates the individual plant's centers within the image.
[image]

joevo2, 2017-01-11:
Wow awesome looking! We were looking into this code of yours https://github.com/FarmBot-Labs/plant-detection to modify it for the tray seeding function. It seems like there will be quite a lot of work for us to customise and learn to build a tray seeding function 😢

joevo2, 2017-01-11:
Any tips on where we could start with building the tray seeding function? Since it is pretty similar with the weeding functionality. Any code for us to reference? Sorry for the trouble again and super thanks for the hard work being done. We would love to help out. Btw me and @menghong777 is actually on the same team in case you wonder :)

roryaronson, 2017-01-11:
The easiest way for you to accomplish tray seeding is to make a hardware jig that allows you to position your tray precisely in the FarmBot workspace. Then you could just use the regular sequence builder to load the various tray sockets with seeds.

If you're set on a camera based solution then you'll need to identify the z-height, X/Y position, and angle of the tray. You would do this by having easily identified markers on the corners of the tray. Then your code would output socket locations (coordinates) which your "seed loading" sequence would need to use.

We were just discussing today how we plan to allow not only "Plant", "Weed", and "Tool" objects in the database, but also any other type of object with coordinates and a label. This is a little far off I think, but when such a feature is developed your code would be able to create "Socket" objects in the database which then your Sequences could import coordinates from. Hope that makes sense. This is basically how weed detection puts the weeds in the db.

ConnorRigby, 2017-01-17:
@joevo2 @menghong777 Starting with release `2.1.9` we added support for other systems. You should download the corresponding `.img` file and flash it to your SD card.

closed 2017-01-17 by ConnorRigby