FarmBot/farmbot_os #235: Lack of /state directory in the root file system

wuulong, 2017-01-22:
### Step 1: Are you in the right place?
This repository is for Farmbot OS problems.

### Step 2: Describe your environment

  * Device: rpi3
  * Farmbot OS version: 3.0.1 ( code version: 1231 )
  * Arduino Firmware version: none

### Step 3: Describe the problem:
While waiting my Farmbot, I try to build the image that on RPI3.
Issue1:
The root file system seems lack of /state directory.
I try to create /state directory in the root file system. Seems fix the issue.
hint: the test environment did not attach Arduino, keyboard, mouse. 
Attach 2 console log in the zip file.

1. console_1.log : the original console output of my build image. It show CRASH REPORT
2. console_2.log: the console output of my another build which include /state in the root file system. Boot complete and can type some commands.

Issue2:
The HDMI seems have no output of my build image. The same test environment with 3.0.0.img in the SD can show some texts in the HDMI.
#### Steps to reproduce:

  1. It always have this issue from clean build image

#### Observed Results:

  * What happened?  This could be a description, log output, etc.

#### Expected Results:

  * What did you expect to happen?
Maybe someone find the /state directory should be added in the build process.

[console_log.zip](https://github.com/FarmBot/farmbot_os/files/721834/console_log.zip)
 
Thanks for your effort, and great to have Farmbot project!

ConnorRigby, 2017-01-23:
Ah yeah this is git not wanting to keep that dir because its empty. Will fix shortly.

ConnorRigby, 2017-01-23:
@wuulong
if you want to fix now, you can just simply 
create a dir called "state" in `ls apps/farmbot/config/hardware/rpi3/rootfs-additions/`

closed 2017-01-24 by ConnorRigby

wuulong, 2017-01-24:
Thanks. That's what I need for now!