FarmBot/farmbot_os #243: Raspberry looses connectivity with web after awhile

fluffy, 2017-02-05:
Things will all be working fine, then all of a sudden the web stuff will not be able to talk to the PI any more. The last messages that come up seem to be stuff about dhcp refreshing. I suspect that this might have to do with when a DHCP lease refresh happens that instead of a refresh it somehow takes the wlan interface up and down. Then the software on PI fails to forma  new connection to web side. I have not tried to debug this much but if people have hints for me, I might try next weekend.

RickCarlino, 2017-02-06:
@fluffy I just noticed this one recently also. We've been noticing some strangeness somewhere in the network stack but have not been able to pin it down yet.

If you do see anything while debugging, please let us know. I'm going to leave this one open until we find a fix.

ConnorRigby, 2017-02-08:
Yeah the DHCP refresh is harmless. Its just usually the last message in the logs to appear. The problem actually lies in the MQTT client. (`gen_mqtt.ex` and `gen_mqtt_client.ex`) The client gets restarted when it dies, but its not actually dying, its just going into some sort of "limp" mode. checking out today

ConnorRigby, 2017-02-08:
Im testing a fix for this. Will report back when i know if it works or not.

ConnorRigby, 2017-02-14:
I believe this is fixed!

closed 2017-02-14 by ConnorRigby