FarmBot/farmbot_os #247: Version 3.0.4

Daniil145, 2017-02-15:
Version 3.0.4

Trouble connecting to bot

older versions normal connect and config

ConnorRigby, 2017-02-15:
Have you tried rebooting your bot and refreshing the web application? Do you see a configurator wireless ssid?

Daniil145, 2017-02-15:
Yes. Reboot+ wifi see connect go 192.168.24.1 downloading page and after one sec "Trouble connecting to bot" 3.0.3 work good

ConnorRigby, 2017-02-15:
so version `3.0.3` works then?

Daniil145, 2017-02-15:
last 3.0.3 work - clean install
3.0.4 work only update on api and dont work on clean install

ConnorRigby, 2017-02-15:
interesting. checking it out right now! thanks

ConnorRigby, 2017-02-16:
This should be fixed next release!

ConnorRigby, 2017-02-17:
I lied. its still broken.

ConnorRigby, 2017-02-27:
this was finally solved with a revert of the `tzdata` package.

closed 2017-02-27 by ConnorRigby