FarmBot/farmbot_os #272: I updated for farmbot-rpi3-3.0.8-rc.6.img and receive this error every time

Daniil145, 2017-03-13:
[image]

ConnorRigby, 2017-03-13:
Can you look on the Devices tab of the frontend, and tell me the `Controller Version`?

Daniil145, 2017-03-13:
[image]
Hmm I use Disk 32 imager and write flash

ConnorRigby, 2017-03-14:
It appears you are still on an old version. For a quick fix to sequences, you can just hit the `SYNC` button then your sequences should be able to be run. 

i pushed a new update last night [herer](https://github.com/FarmBot/farmbot_os/releases/download/v3.0.8-rc.7/farmbot-rpi3-3.0.8-rc.7.fw) Could you try that one?

closed 2017-03-20 by ConnorRigby