FarmBot/farmbot_os #281: Support for WPA2 EAP Wi-Fi authentication

TheAmazingAussie, 2017-03-29:
Hello,

So I am a part of a group of people who are helping to create this project, it's taking place at a university, and we would like to use the on-site Wi-Fi available.

The issue is that our Wi-Fi authentication system is WPA2 EAP (where you use your student credentials which is the email and password) to authenticate your device, and within the initial setup of the FarmBot, to be able to sync to the online FarmBot account, it only has one input option which is for a password only.

So, in short.

- The initial setup only has one input box for a password only
- We need to use both our email and password for authentication, which would be two input boxes.

So is this already supported, or a workaround? If not, would it be possible to be supported?

Thank you

ConnorRigby, 2017-03-29:
So it is currently not supported and no _easy_ workaround that i can think of. I will investigate adding support for this.

ConnorRigby, 2017-03-29:
@TheAmazingAussie  I am looking for a bit more information here. When you configure your wifi on other devices, you just give a id and password right? do you need to provide a certificate?

TheAmazingAussie, 2017-03-30:
Yes, just a id (email) and password, no certificate required (as far as I'm aware?), since it's a bring your own device sort of thing. My phone and laptop are able to connect without needing a certificate, just credentials required.

For example, this is from their 'connection guide' at my university (yes, it's Windows but it should still give you an idea of the type of system).

http://i.imgur.com/1f6tw0Q.png 

http://i.imgur.com/6iR7HeP.png

http://i.imgur.com/yZ1zTNy.png

Does that help at all?

ConnorRigby, 2017-03-30:
Yes that is helpful. I will take a look in the morning

TheAmazingAussie, 2017-04-05:
Hey Connor, has there been any progress? 

I was thinking it might be pretty tough for you to find a router that has WPA EAP support to test this, is that correct?

ConnorRigby, 2017-04-05:
@TheAmazingAussie im checking this right now actually! I have an extra router with DD-WRT on it that im trying to setup with WPA2 EAP right now.

vaibhavgupta3110, 2017-04-05:
not sure if this is helpful but I remember for Enterprise networks (recently set one of my own up for my office) a RADIUS client should be enough. Do any hex packages exist which can onboard a RADIUS client?

ConnorRigby, 2017-04-05:
i know of  [this](https://github.com/travelping/eradius) which has a radius client in it. Never used it before tho.

ConnorRigby, 2017-04-05:
@TheAmazingAussie So my router doesnt have support for PEAP authentication. Still investigating.

TheAmazingAussie, 2017-04-05:
Hmm, alright. Thanks for keeping me updated.

roryaronson, 2017-04-05:
@connorrigby, we can order you a router with the right enterprise specs if you need it to get this working. @TheAmazingAussie do you know which router your network uses?

TheAmazingAussie, 2017-04-06:
@roryaronson No idea sorry, but do these links help?

https://www.eduroam.edu.au

http://www.usc.edu.au/learn/student-support/information-technology/wireless-on-campus

TheAmazingAussie, 2017-04-18:
Just wondering, is there any other way for this to connect to the internet?

Such as using an ethernet cable or a USB with a paid plan.

roryaronson, 2017-04-18:
Yes you can use Ethernet - no paid plan required :)

TheAmazingAussie, 2017-04-19:
Alright, I guess I can use Ethernet as a workaround. I won't be closing the ticket just yet as I won't be able to test it _just_ yet.

ConnorRigby, 2017-04-19:
Yes please don't close the issue, it' is a much needed feature.

ConnorRigby, 2017-05-01:
@TheAmazingAussie 
Hello, just want to let you know, I haven't forgotten about this issue. I just haven't had time to look into it yet.

TheAmazingAussie, 2017-05-01:
All good, we believe we have a workaround which can satisfy us. Like I said before, the ticket won't be closed as I know you said this is a feature which needs to be added.

Take your time, there's no urgency for this feature at the moment.

Klimbim1, 2017-06-10:
I noted today that the bot disconnected once from my WLAN and after power down and up again there are each 2nd time problems to connect to the WiFi. 

Please let me know how to deliver you more information in order to debug, that is completely annoying...

ralphwu, 2017-11-01:
I solve it by using OrangePi Zero with OpenWRT/LEDE installed as a bridge of LAN and Wifi interface

You can get the install image at http://pulpstone.pw/sunxi/ 

Remove the wpad-mini package and install the wpad package then WPA2-EAP with be supported.

ConnorRigby, 2019-01-22:
Should be fixed in v7.0.0!

closed 2019-01-22 by ConnorRigby