FarmBot/farmbot_os #282: End stops don't work on after new release

Daniil145, 2017-04-03:
my.farmbot.io
You disable endstops?

ConnorRigby, 2017-04-03:
@Daniil145 sorry about that, they werent disabled, but there was a new version of teh Arduino Firmware released. 
Could you try disabling and re enabling end stops?

Daniil145, 2017-04-03:
@ConnorRigby I dont understand "disabling and re enabling end stops" on interface dont have toggle off and on Endpoints. What can I do to activate them? With the loss of the Internet (especially when I use mobile Internet) the system behaves unpredictably. Especially this is critical for the Z axis. To do this, I need limit switches.

ConnorRigby, 2017-04-03:
Oh sorry i didnt realize they werent on the interface any more. 
@RickCarlino any thoughts on this one?

RickCarlino, 2017-04-03:
@Daniil145 I've recently added endstop toggle buttons to the device panel. They will be available in the next production release (probably this week).

closed 2017-04-03 by RickCarlino

RickCarlino, 2017-04-04:
@Daniil145 We just did a production deploy. Please let us know if you find any issues.