FarmBot/farmbot_os #283: No wifi configurator appears

issamsham, 2017-04-05:
### Step 1: Are you in the right place?
This repository is for Farmbot OS problems.

### Step 2: Describe your environment

  * Device: RB pi3
  * Farmbot OS version: 3.0.8
  * Arduino Firmware version: _____

### Step 3: Describe the problem:

after flash my SD card I could not find any wifi connection by my RB pi3
#### Steps to reproduce:

  1. _____
  2. _____
  3. _____

#### Observed Results:

  * What happened?  This could be a description, log output, etc.

#### Expected Results:

  * What did you expect to happen?

ConnorRigby, 2017-04-05:
I could use a little more information. 
* How did you flash the sdcard? 
* Whats your power supply rated at? 
* What devices have you tried searching for configurator?
* did you download the correct image for your device?
* what are the leds on the raspberry pi 3 doing during boot?

issamsham, 2017-04-07:
- I flashed my SD card by using Etcher 1.0.0 beta also I have tried to flash it by using Terminal both did not work, 
- I am using Samsung USB power adapter 5.0 V--- 2.0 A. 
- I used my iPhone and My mac to search for Configurator wifi Signal. 
- I tried The latest version of Farmbot os Image and Also I have tried the RC version.
- It's Red for the whole time. 
I tried to use a new Raspberry pi 3 also and New SD card.

ConnorRigby, 2017-04-07:
If its red the entire time (with no green) the sd card isnt getting flashed properly. 
just to be sure [this](https://github.com/FarmBot/farmbot_os/releases/download/v3.0.8-rc.8/farmbot-rpi3-3.0.8-rc.8.img) is the image you should be using.

issamsham, 2017-04-08:
The light is red the entire time,
Dear the link does not work

ConnorRigby, 2017-04-08:
sorry about that. [this is the  correct link](https://github.com/FarmBot/farmbot_os/releases/download/v3.1.0/farmbot-rpi3-3.1.0.fw)

issamsham, 2017-04-10:
Hi,
The attached file is not image file so it does not work with Etcher, any suggestions please?

issamsham, 2017-04-10:
I have tried to flash the SD card using Terminal also it doesn't work

dwhickox, 2017-04-10:
This is the correct link for the RPi3 [Link](https://github.com/FarmBot/farmbot_os/releases/download/v3.1.0/farmbot-rpi-3.1.0.img). Are you sure you are using an RPi3, they look similar and may be confused check the silk screen on the boar because just because you ordered and Rpi3 doesn't mean you got one. 

Also instead of using etcher, try win32 disk imager [click here](https://sourceforge.net/projects/win32diskimager/) it is really reliable

One other thing to try is plug in the Arduino into the raspberry pi and see if it works then if it doesn't work unplug it again and try again.

ConnorRigby, 2017-04-17:
@issamsham any update on this?

issamsham, 2017-04-19:
Hello @ConnorRigby, First of all thank you to follow up this issue with me, I am out of the country I will inform you with the new result once I get back, 
Appreciate your effort

xiaoqingrong, 2017-05-07:
I installed it on the raspberry pi3, and the screen was initially red, and then the black screen didn't change.

ConnorRigby, 2017-05-10:
@xiaoqingrong I dont think that issue is shared with this one. I will try to help you in a separate issue.
@issamsham i am going to close this for now. If on the latest version you still get this issue, please reopen.

closed 2017-05-10 by ConnorRigby