FarmBot/farmbot_os #284: Cannot update the MAX SPEED/ACCELERATE FOR STEP/TIMEOUT

bdudout, 2017-04-08:
* Farmbot OS version: Version3.0.8-rc.8 (latest currently)
  * Website tested : both my.farmbot.io and staging.farmbot.io

### Step 3: Describe the problem:

In the form DEVICE > HARDWARE it is only possible to update STEPS PER MM and the toggle buttons.

The following fields are not able to update the value : MAX SPEED/ACCELERATE FOR STEP/TIMEOUT

#### Steps to reproduce:

  1. Connect to the website (either my or staging)
  2. Go to DEVICE > HARDWARE
  3. Update values for MAX SPEED/ACCELERATE FOR STEP/TIMEOUT
  4. Save

#### Observed Results:

The values are going back to their previous value (not especially the default one, the previous one)

#### Expected Results:

Values should update (working for the steps per mm)

Do not hesitate to ask me for more testing !
Thanks :)

dwhickox, 2017-04-10:
This has been a known issue for quite a while, I have been having this problem for a long time, as far as I am aware they are currently working on implementing all the buttons and settings into the software, but the web interface was developed long before the functionality was there. So this might not be a "bug" since I don't know if there is even code in the OS to make it work, but I am a couple of weeks behind on their progress. If it has been "fixed" then count me in on it doesn't work for me either.

ConnorRigby, 2017-04-11:
Do the toggle buttons work? 
I think this might be a bug in two places.
Investigating today.

bdudout, 2017-04-11:
Yes, toggle buttons are ok :)

ConnorRigby, 2017-04-14:
Sorry we havent had time too look more into this in full. I believe i know what is going on tho.

ConnorRigby, 2017-04-17:
@bdudout i believe this to be fixed in our staging environment. Should be up soon on production.

ConnorRigby, 2017-05-01:
@bdudout hopefully you can confirm this is fixed now.

closed 2017-05-06 by roryaronson