FarmBot/farmbot_os #287: i can run a OS on raspberry pi

LadZone, 2017-04-11:
To enable us to help you with the issue, please provide as much information as possible about your device and the problem. At a minimum, provide:
### Step 1: Are you in the right place?
This repository is for Farmbot OS problems.

### Step 2: Describe your environment

  * Device: raspberry pi2
  * Farmbot OS version: farmbot-rpi2-3.0.8.img
  * Arduino Firmware version: not sure

### Step 3: Describe the problem:

#### Steps to reproduce:

  1. its just doesnt bootup the raspberry pi
  2. _____
  3. _____

#### Observed Results:

  * What happened?  This could be a description, log output, etc.

#### Expected Results:

  * What did you expect to happen?

ConnorRigby, 2017-04-11:
I believe you are the first one to test Raspberry pi 2, so thank you!

Just a note but you will _NOT_ see a wireless ssid on any other device except raspberry pi 3 making it difficult to configure. Expected behavior should be that your raspberry pi 2 is to be plugged in via Ethernet and it will automatically get an ip address, you can then configure your bot by navigating to its ip address in a web browser. 

so to troubleshoot:
1. What image did you flash onto your device?
2. What do the LED's do when you power the board up?
3. You have dhcp network access right?

Just one last note:
Support for devices other than Raspberry Pi 3 is minimal, we strongly suggest picking up a Pi3 as it is much easier to configure, and runs 4 times faster.

LadZone, 2017-04-11:
[image: Inline image 1]
What do the LED's do when you power the board up?
For LED it just flash and turn off the whole system

You have dhcp network access right?
and yes i do have dhcp network

roryaronson, 2017-05-06:
Because we are focusing on developing new features right now, we are temporarily not supporting other platforms such as rpi2, 1, or 0. We anticipate bringing support back in the future and welcome pull requests, but in the meantime we encourage people to get a pi3.

closed 2017-05-06 by roryaronson