FarmBot/farmbot_os #291: Allow forced update to final release version or rollback of OS

ivanmayes, 2017-04-15:
It would be ideal to be able to roll back from release candidate builds so that testing can happen on bots and roll back to stable afterwards.  Since most people only have one Farmbot, and the SD card is not easy to access from the casing, would be great to have a little more control over this.

ConnorRigby, 2017-04-15:
It is not expected to be able to move to and from release candidate builds. They are intended for developers only for testing new features; not long term use. 

There is however a way to manually update to a later version. although you will not be able to go back to a staging build.

0. Download the wanted `farmbot-<target>-<version>.fw` 
   * NOTE: Make sure you download the `.fw` file, not a `.img` file. It should be about 56 megs, not three hundred something.
0. Navigate in a web browser to `http://<your_ip_address>/firmware/upload` 
0. Upload that file to your bot. It will take a couple minutes, but should come back up shortly.

Please note this is not intended for a production or long term solution. It is not guaranteed to work every time, and is disabled on production builds in the future.

closed 2017-04-15 by ConnorRigby

ivanmayes, 2017-04-16:
Hey Connor!  I was speaking from a position of development, but I wasn't aware of the firmware upload page, thanks!