FarmBot/farmbot_os #292: Peripheral Pins are not properly turned on

AscendFB, 2017-04-16:
### Step 2: Describe your environment
  * Device: Raspberry Pi 3
  * Farmbot OS version: Version3.1.2
  * Arduino Firmware version: -

### Step 3: Describe the problem:
When I turn a Pin for peripherals on, like Pin 9, the LED on Ramps shines really dark. I can only measure 0,3V when its turned on. This affects all pins that i've tested.

#### Steps to reproduce:

  1. Use current OS Version 3.1.2
  2. Add a Pin for Peripherals at my.farmbot.io/app/controls.
  3. Turn the button to "on" and take a look at the Ramps LEDs.

#### Observed Results:

** I knew that it was working recently, so I tested this with an earlier OS Version, Version 3.0.8-rc.8.
Everything worked fine, LEDs on Ramps were much brighter and there was +12V at the pin output when turned on.  **

#### Expected Results:

There should be  +12V at the selected pin connection.

Btw: I also tried to fix with a different Ramps and a different Arduino . Both didnt change anything.

ConnorRigby, 2017-04-16:
Hm. We've had some changes to the Arduino firmware in the last couple
releases. I'll take a look into it.

ConnorRigby, 2017-04-17:
@AscendFB ok sorry about that. I fixed it and am building a release right now.

closed 2017-04-17 by ConnorRigby