FarmBot/farmbot_os #302: raspberry pi

xiaoqingrong, 2017-05-05:
Will the farmbot_os system be installed on the raspberry pi? Because of the black screen。I don't know what I should do.

RickCarlino, 2017-05-05:
@xiaoqingrong Have you tried to [configure the app over wifi](https://software.farmbot.io/v1.0/docs/wifi-configurator) ? If FarmBot OS installed properly, you should see a `farmbot-****` wifi network available.

xiaoqingrong, 2017-05-06:
[image]
Thank you.No, I did it, right?

xiaoqingrong, 2017-05-08:
The computer is always black.WIFI devices can be found in farmbot__xxx.Connect to the WIFI and open the web page at http://192.168.24.1/ and the computer displays the blue screen.I don't know what to do next

ConnorRigby, 2017-05-10:
closing due to duplication. 
https://github.com/FarmBot/farmbot_os/issues/305

closed 2017-05-10 by ConnorRigby