FarmBot/farmbot_os #305: The system configuration

xiaoqingrong, 2017-05-08:
To enable us to help you with the issue, please provide as much information as possible about your device and the problem. At a minimum, provide:
### Step 1: Are you in the right place?
This repository is for Farmbot OS problems.

### Step 2: Describe your environment

  * Device: windows
  * Farmbot OS version: 1.3.5
  * Arduino Firmware version: Genuino MEGA 2560

### Step 3: Describe the problem:
The computer is always black.WIFI devices can be found in farmbot__xxx.Connect to the WIFI and open the web page at http://192.168.24.1/ and the computer displays the blue screen.I don't know what to do next

#### Steps to reproduce:

  1.Follow the official farmbot_os system installation method_
 
#### Observed Results:

  * What happened?  This could be a description, log output, etc.

#### Expected Results:

  * What did you expect to happen?

RickCarlino, 2017-05-08:
@xiaoqingrong Which browser are you using? We only support the latest version of Chrome and FIrefox at the moment.

Also, can you please show us a screen shot (or copy paste) the contents of the Javscript console?

You can access it on windows using `CTRL + SHIFT + J`.

ConnorRigby, 2017-05-10:
Im having trouble understanding the issue.
When you get to the "blue screen" 
does it look like this?
[image]

If not could you take a screenshot of what you are seeing?

ConnorRigby, 2017-05-29:
@xiaoqingrong not sure if there is still an issue. Please re-open if you have any problems.

closed 2017-05-29 by ConnorRigby