FarmBot/farmbot_os #314: Logging of messages

Klimbim1, 2017-05-31:
To enable us to help you with the issue, please provide as much information as possible about your device and the problem. At a minimum, provide:

### Step 2: Describe your environment

  * Farmbot OS version: latest 3.1.6

### Step 3: Describe the problem:

Please let me know how to access log files on the raspi;
I want to debug why the farmbot os does only connect to my WLAN and not to my local server;
therefore I need to read logfiles of it; how and where can I do that?

ConnorRigby, 2017-05-31:
Log files can be fetched by plugging a fat32 formatted USB drive into the bots usb port.

Not 100 percent sure, but i have a suspicion you need to pass the `-b` flag to your rails server.

Klimbim1, 2017-05-31:
Does not work with the FAT32 formatted usb stick... is there any convention on naming the USB stick or something like this?

-b flag in order to do what exactly?

RickCarlino, 2017-05-31:
@Klimbim1 This was recently (past hour) updated in the README for the FarmBot Web API.

When running a Rails server via `rails s`, passing a `-b` flag will open the port to *all* interfaces rather than just `localhost`.

ConnorRigby, 2017-06-01:
@Klimbim1 also in regards to your original question the best way to see log messages is to get yourself an FTDI cable such as this one [here](https://www.digikey.com/product-detail/en/TTL-232R-RPI/768-1204-ND/4382044?WT.mc_id=IQ_7595_G_pla4382044&wt.srch=1&wt.medium=cpc&WT.srch=1&gclid=Cj0KEQjw9r7JBRCj37PlltTskaMBEiQAKTzTfKNgtoiiMcV0yjgL0WnPQygi6civ3NGg-IbNqEHhcOoaAjch8P8HAQ)

also to the logs thing. You have to plug the usb stick in while the bot is running. (not before you boot it up)

Klimbim1, 2017-06-10:
Could you please add the time on the log file which is written on the USB stick? That is so important....

Klimbim1, 2017-06-10:
How would I get the log messages then, @ConnorRigby with the FTDI cable? Is the mqtt server the same log stream?

ConnorRigby, 2017-06-12:
FTDI is as simple as just accessing the tty. For example in Linux i issue a command among the lines of
`screen /dev/ttyUSBCHANGEME 115200`
Mqtt will also display logs to the shell i believe, but not filtered logs iirc.

as for time in the logger i hadnt planned on writing time, but i might be able to put a unix timestamp in there.

ConnorRigby, 2017-06-12:
Closing.
If you have further questions we ask that you post them on the Forum.

closed 2017-06-12 by ConnorRigby

Klimbim1, 2017-06-26:
Hey guys,
could you pleeeeease insert a date stamp with the time stamp and not cut the rollbar log after some xx lines???? You could have a setting in the device page where one could select only view the last 100 lines or even everything... quite annoying if the bots tries to get new tokes one thousand times a day and you cannot see if it did the movement yesterday evening or not...