FarmBot/farmbot_os #315: Double Axis movements when moving only one axis

Klimbim1, 2017-05-31:
### Step 2: Describe your environment

  * Farmbot OS version: latest 3.1.6
See settings:
[image]
[image]

### Step 3: Describe the problem:
When setting 1 mm of step width and pressing the z movement (up in my case) the x axis moves approx. 1 mm too. This is happening abitrarily and not reproducible, thus statistically distributed;

Did not encounter it with the 10 mm setting...

ConnorRigby, 2017-05-31:
This sounds like an issue with the arduino-firmware.
@TimEvWw care to chime in?

Klimbim1, 2017-05-31:
If i would know how to leave you log messages of the encoders you debug more in detail... :/

bonanzajellydog, 2017-05-31:
I have also experienced this issue: movement on X when I only wanted to
move Z. I have encoders disabled but otherwise my settings are the same as
above and also 3.1.6.

Klimbim1, 2017-06-05:
@TimEvWw & @ConnorRigby any more news? How can I help you to debug...?

ConnorRigby, 2017-06-05:
I am unable to reproduce (because my farmbot only has about 100 mm movement in any direction).
I will look into this later this week when i have access to a full size bot.

Klimbim1, 2017-06-06:
Hi Connor,

that shouldn't be of a matter since the error occurs when moving 1 mm with the bot?!

ConnorRigby, 2017-06-06:
Very interesting. On both my small farmbot and on a deconstructed farmbot I don't notice this at all.

Does it happen during sequences or during manual control?

Klimbim1, 2017-06-06:
Manual control. Sequence I did not try yet...

dwhickox, 2017-06-08:
@Klimbim1 are you sure you plugged the second x motor into the correct slot, not the second z motor connection on accident? (based on your description this is doubtful but I have done dumber things) Also try enabling always power motors on all the axises (in settings) because if the x motor shifts to somewhere in between a step while it is not powered it will move when powered up, and if I am correct when any axis moves all three axises are powered so that may cause the slight jump issue you are experiencing, and that would also be the best reason for its intermittents.

Klimbim1, 2017-06-10:
Hi @dwhickox,
I don´t get your first idea, which second Z motor?

Power on motors always on does not solve this bug. I experienced it today in a sequence as well, so it is not related to manual control. Please try solving this, otherwise one could crash into the toolbay if picking up seeds or the like...

roryaronson, 2017-06-10:
A little update here: we were able to reproduce the issue yesterday. As @Klimbim1 mentioned, it does not happen consistently. The issue appears to be in the Arduino firmware, where it is keeping track of the locations. The bug seems to happen in this example circumstance:
1) Tell the bot to move to (0, 80, 0)
2) The bot attempts to move to (0, 80, 0) but only makes it to (0, 78, 0) for some reason
3) Tell the bot to move along the X axis some amount, say 20mm.
4) The Arduino is computing the new movement as originating from the last position that it was supposed to go to (0, 80, 0) rather than from the last position it was actually at (0, 78, 0)
5) This results in a small movement of 2mm along the Y axis when the X axis is moved. So from (0, 78, 0) to (20, 80, 0)

There are several things that we're working on to resolve this:
1) Figure out why FarmBot is not reaching the 80 position.
2) Ensure that movements are computed as originating from the current position rather than the last position it was supposed to go to.
3) Implementing a "retry movement" function that will be useful to overcome stalls/stutters so that FarmBot can try again to reach its destination
4) Improving the "Use Encoders for Positioning" feature

Klimbim1, 2017-06-10:
Hi Rory,

what I thought today and which is fitting into that context:
Why are the encoders not counting the position if you disable power on the motors (in the settings) and you then move the bot by hand....? I am really wondering about the whole positioning and keeping up with it concept of the bot... there does not seem to be a sophisticated one : p

TimEvWw, 2017-06-14:
The setting for counting position using the encoder is not on the screen yet. The simple reason is that it sometimes misses a step if that step is just a bit too fast. After some time those errors start accumulating into larger deviations.

Moving to 0,78,0 while the target is 0,80,0 can happen when the bot detects it was getting stuck using the encoders. If the next command is to *,80,* it then tries again to go to go to that position.

Klimbim1, 2017-06-26:
Is the double axis movement solved? Do you want to keep this issue open for the implementation of Tims idea of position counting?

Klimbim1, 2017-08-07:
Hey guys, two month are gone now and no reaction, C´MON!!!!

Klimbim1, 2017-09-10:
Any more news on that? Did you find the fault? Did you solve the bug? Anything???

ConnorRigby, 2017-09-11:
@Klimbim1 Are you still hitting this bug? I had assumed it was fixed in the Arduino Firmware. I personally don't hit this bug with my configuration.

closed 2018-01-10 by ConnorRigby