FarmBot/farmbot_os #316: Tool verification read pin

Klimbim1, 2017-06-05:
To enable us to help you with the issue, please provide as much information as possible about your device and the problem. At a minimum, provide:
### Step 1: Are you in the right place?
This repository is for Farmbot OS problems.
I don't know if this should be covered here or in the web-api, please move if so...
### Step 2: Describe your environment

  * Device: what do you mean?
  * Farmbot OS version: 3.1.6
  * Arduino Firmware version: where can I read that?
### Step 3: Describe the problem:
If I execute this to read the tool verification pin:
[image]

It always says: ... set PIN: 63: 1, if I measure between B and C there is no 5V..., even during the execution of this sequence...

RickCarlino, 2017-06-05:
This was fixed on a recent version of FarmBot OS that has not been released yet. Please check the releases page in a week or so.

closed 2017-06-05 by RickCarlino

Klimbim1, 2017-06-26:
Does not work at the moment, it always says 1, regardless if I have a tool mounted (with installed jumper) or not... please give information on the best way to debug....