FarmBot/farmbot_os #317: SSH account with root and no password doesn't work

siteswapjuggler, 2017-06-07:
* Farmbot OS version: latest

### Step 3: Describe the problem:

SSH is activated (I saw the port open when I scan the network) but I do not succeed to connect with the account information given in the FAQ (root/"no password").

#### Steps to reproduce:

  1. enable SSH during config
  2. reboot
  3. try to connect via putty : root@farmbotipaddress, no password

#### Observed Results:

  * The SSH connection is not allowed.

#### Expected Results:

  * SSH connection to be establisehd and access to a root terminal

ConnorRigby, 2017-06-12:
What exactly are you looking for that requires SSH access?
I tend to discourage SSH, because there is no proper Linux shell, package manager, or anything else really.

ConnorRigby, 2017-06-13:
Ok i fixed SSH. Will be available again in the next release.

closed 2017-06-13 by ConnorRigby

siteswapjuggler, 2017-06-18:
Thanks a lot :) I actually got no precise need for myself but I manage a studend who would like to explore the system from the inside for better understanding. She was just surprised not to connect with the actual instruction.

Thanks a lot for your answers. Looking forward the next release :)

rootux, 2017-09-19:
I need to activate the pi camera module and I need to run `raspi-config` yet can't enable SSH on the latest version

gabrielburnworth, 2017-09-19:
@rootux Are you using the web app? You can just select `Raspberry Pi Camera` in the camera selection dropdown in the __Device__ widget.

FarmBot OS is not Raspbian, so there is no `raspi-config`.

rootux, 2017-09-19:
@gabrielburnworth I'm using the web app.
Bought a pi2 camera module. Plugged it in.
When I select the `Raspberry Pi Camera` from the web app it's not working - no picture is shown

gabrielburnworth, 2017-09-19:
@rootux I don't want to get too off-topic for this issue. Can you create a new issue including 
the message you get when you click `Take Photo`?

rootux, 2017-09-19:
@gabrielburnworth I did - But it does not show any error but "Farmbot [take-photo] Problem getting image." - Is there a better way to get the logs?
Here is the issue https://github.com/FarmBot/farmbot_os/issues/349