FarmBot/farmbot_os #318: Building probs

Klimbim1, 2017-06-10:
```
~/farmbot$ cd os/
klimbim@ubuntu:~/farmbot/os$ mix deps.get
Env
  MIX_TARGET:   rpi3
  MIX_ENV:      prod

No Buildroot dir found!
using rpi3 - prod configuration.
** (Mix) Archive "nerves_bootstrap" could not be found. Please make sure the archive is installed locally.
klimbim@ubuntu:~/farmbot/os$ npm install nerves_bootstrap
npm ERR! code E404
npm ERR! 404 Not Found: nerves_bootstrap@latest

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/klimbim/.npm/_logs/2017-06-10T23_44_21_132Z-debug.log
```
What can I do?

ConnorRigby, 2017-06-12:
# TLDR
couple things here.
* You need to install `nerves_bootstrap` as per these [here](https://hexdocs.pm/nerves/installation.html)
* you should just `npm install` not `npm install nerves_bootstrap`

# Further reading.
* `nerves` is an Elixir project/framework/dependency.
* Farmbot OS uses a few different languages and frameworks that you will need to have set up.
   * `nodejs` version `7.0.0`
   * `typescript`version `2.1.4` - which will get installed via `npm install`
   * Erlang version `19.3`
   * Elixir version `1.4.2`
   * Nerves Framework version `0.5.1`
   * Nerves bootstrap version `0.3.1`

I wrote a small post about managing version and stuff like this on embedded Elixir [here](http://embedded-elixir.com/post/2017-05-23-using-asdf-vm/)

closed 2017-06-12 by ConnorRigby

AscendFB, 2017-06-15:
Hey @ConnorRigby 
I also ran into trouble recently when I was trying to make a build. Its not important for me, but maybe I did something wrong in the process of setting up the environment.
Here is the last part of the terminal log:
[image]

I am using ubuntu 16.10 via Virtual box.
My versions:
nodejs: 8.1.2
Erlang: 19.3
Elixir: 1.4.2
Nerves: ? ->installed as mentioned in building.md guide.

I also tried using the newest Erlang/Elixir versions, but didnt change anything. Any idea whats happening there?

ConnorRigby, 2017-06-15:
It's looks like you might need Ruby to be installed