FarmBot/farmbot_os #320: WIFI connection & reconnection

Klimbim1, 2017-06-13:
### Step 2: Describe your environment

  * Device: kit as sold
  * Farmbot OS version: 3.1.6  
  * Arduino Firmware version: as in 3.1.6

### Step 3: Describe the problem:
WIFI connection errors:
WIFI connection statistically distributed interrupts, currently no way of logging it / debugging it and seeing when what why?! Disconnection interval rather some hours than minutes or seconds.
WIFI does not reconnect after power down or loosing connection as per above written. No way of debugging it as per above written.
Very strong WIFI signal, shouldn´t be a problem of coverage.

How is internet connection coded? Is it sending alive signals at fix intervals? Is there a constant connection? What causes exactly reboots of the bot (how long is a disconnect allowed etc.) once WIFI connection is established?

ConnorRigby, 2017-06-13:
# TLDR 
There is currently an error in the logic in the code that handles WiFi connectivity. It has since been fixed but not released yet. (we are still QA'ing the release before public consumption).

# Detailed explanation
Basically low level linux calls were not guaranteed like i thought they were, so in the particular event of `CTRL-EVENT-DISCONNECTED` (a message that comes from low level WiFi handling code) i assumed that the access point had gone down. But in the real world things like radio interference and timeouts cause this message to appear. It has since been fixed by setting a retry. So when this event happens it is not always treated as a error situation, and handled accordingly. 

## Extra reading 
if you are interested in the code that handles network, most of the hard work goes on [here](https://github.com/nerves-project/nerves_wpa_supplicant)

the more *high level* code that handles it from farmbot os is pretty simple too.

We have been playing with ideas of having certain rules for stuff like this. for example: 
* number of retries
* timeouts
* what to do in the case of the access point not being found etc

I'll leave this issue open since the new release is not out yet.

Klimbim1, 2017-06-14:
Hi @ConnorRigby,
as that presents a great security risk: one could log into the OPEN Wifi, create a Farmbot account before (no check wether you bought a device or not!!) and overtake the control of my farmbot, this needs to be fixed ASAP.

Does that bug explain as well why the bot does not log in automatically after power off or loosing the wifi connection?

ConnorRigby, 2017-06-14:
Yeah the bug where farmbot doesn't come back online after reboots sometimes has also been fixed. It was of the same nature.

Klimbim1, 2017-06-14:
Has been fixed means in the new release?
Currently my bot is not usable because it disconnects after some minutes and does not reconnect then... :(

ConnorRigby, 2017-06-14:
The release is planned for the next few days.

Arduns, 2017-06-19:
@ConnorRigby  when will publish the new release

ConnorRigby, 2017-06-19:
@Arduns we plan on having the release out later today.

Klimbim1, 2017-06-19:
Hi @ConnorRigby,

last friday, Ascend build the OS 4.0.0.; I installed it and tried it yesterday via the staging server;

At the moment during every second movement with encoders on, the bot issues a emergency message because the encoder detected a stall and then simply stops; If that problem is solved in the new build, consider this message as not sent; if not the bot will not be usable;

In the frontend logbar you can only see the last xx messages. Therefore you cannot see anywhere the whole log; this would be nice to change too, because the bot is flooding with ntp updates and stuff which isn´t really useful; moreover does the bot log when it lost WIFI?

Cheers
klim

ConnorRigby, 2017-06-19:
> At the moment during every second movement with encoders on, the bot issues a emergency message because the encoder detected a stall and then simply stops; If that problem is solved in the new build, consider this message as not sent; if not the bot will not be usable;

You will need to increase your `max missed steps count`. All bots/environments are different so this setting needs to be tuned on a per bot case.

The bot won't/can't log if wifi (internet) connectivity is lost. It holds onto messages until network connectivity is reestablished but the messages are not view-able until the bot has internet. We have been trying to determine a way to best handle that but haven't put much work into it as of yet.

Klimbim1, 2017-06-24:
Still same problem with 4.0.0; its seems more stable but problem is still not solved:
Something caused a factory reset last time. Here's what I know:

I could not find the wifi access point. Check that it was inputted correctly. I tried 6 times and still found nothing. Maybe I'm not close enough to the access point?

As you never offer any proper debugging log files I don´t know how to debug. That is really annoying guys!!!!!!!!!!!!

ConnorRigby, 2017-06-24:
Do you have a intermittent or distant wifi access point/router? Farmbot isn't able to find it. Another concern might be the power supply. What's your power rated at?

As for logs, to see them you will need an FTDI cable. That is how you monitor an embedded device (such as farmbot) in most cases. You can also enable ssh to watch your bot but that is subject to network outage also and highly not recommend because it is a security risk. Unfortunately this is standard practice for all Linux powered headless devices. If you have any better suggestions I would love to hear them but if you are interested in the low level logs, for now I suggest you pick up a serial/ftdi cable

Klimbim1, 2017-06-25:
> Do you have a intermittent or distant wifi access point/router?

What do you mean? The router is on 24h and is 10 m away, that cannot  be the cause of the problem.
You can write something to a log file if something goes wrong. Its just a matter of coding.
If you would write more specific information into the logfile, one could see now what the source of the problem was. Did he not find a wifi? Could he not log into it? How many times did he try to? With which username, bla bla bla; It´s great to say he couldn´t find it. That´s the same if you would ask me which car I like and I would answer "red"... I am not interested in them, I am just trying to get all the big bugs nailed down which you didn´t see or did not want to solve...

ConnorRigby, 2017-06-25:
so that message comes from the core Linux utility called `wpa_supplicant` it tried 6 times to find your wifi network, and it could not establish a connection too it. if your interested read the documentation [here](https://linux.die.net/man/8/wpa_cli). Its a security risk too put the username and password in the logs. and for that matter there is no log "file" to begin with. If your interested in the reasoning behind that you should see this [here](http://events.linuxfoundation.org/sites/events/files/slides/fs-for-embedded-full_0.pdf)

This is usually a problem with distance, power, or latency. If you have access to one, try putting a repeater between your farmbot and your router. There are many issues with the WiFi driver written by the Raspberry Pi foundation (called `brcmfmac`) you can read about them [here](https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=138312)

I could make a setting to disable a factory reset if it can't find your wifi network and just retry infinitely, but if your bot is not connected to your network and it tries to do any network operations they will fail. Most things are made to handle this with the exception of one situation. there is a timer that fires every 2 hours to get a brand new token  to communicate over the web api. Check [this](https://jwt.io/introduction/) out for the description  of that. 

I have two bots that are _always_ on and i don't hit this issue so it is difficult for me to make suggestions. "log messages" won't help either because its obvious _what_ is happening, just hard to determine _why_ in your specific situation.

Klimbim1, 2017-06-26:
Hi @ConnorRigby, 
trying to summarize and answer your possible causes:

Distance:
Cannot be a problem because the bot is placed 10 m away from the access point, no way!

Power:
Power supply to the FB??? Shouldn´t be a problem since the thing is connected to the normal "house" electrical network. I don´t know how prone the system is to power variations from the AC/DC transformer? Any hints on a possible problem here?

Latency: 
Should not be any problem because of the statements answered in the distance section.

Ideas: 

- Logging some voltage levels in order so wether they are in a critical low voltage state?

- Trying to get more information from you on the time that there could be a internet connection loss, a wifi connection disruption or something like this; When is the bot monitoring the WiFi connection, does it send a "I am still online, are you (server) too?" each now and then or how does this link monitoring work?

ConnorRigby, 2017-06-28:
> Power:
Power supply to the FB??? Shouldn´t be a problem since the thing is connected to the normal "house" electrical network. I don´t know how prone the system is to power variations from the AC/DC transformer? Any hints on a possible problem here?

The power issue lies on the DC side of the converter. You should give [this](https://www.raspberrypi.org/help/faqs/#powerReqs) a read

> Logging some voltage levels in order so wether they are in a critical low voltage state?
There is no way from software to know what the voltage levels are. You will need to read on your converter. If it is the one supplied via the Farmbot Kit you won't have to worry about this.

> Trying to get more information from you on the time that there could be a internet connection loss, a wifi connection disruption or something like this; When is the bot monitoring the WiFi [...]

This isn't an internet connection issue, its a wifi connection issue. [here](https://www.quora.com/What-is-the-difference-between-a-WAN-and-a-LAN-port) is a good summary of what i am talking about. 
but to try to answer your question: 

  * every two hours the bot will try to get a new JWT from the API
  * mqtt exchanges a ping/pong message back and forth constantly. (as defined in the MQTT spec)

so as long as your bot is connected to the WAN, you shoudn't have to worry. What is happening to _your_ device is it is disconnecting from the LAN (and thus the WAN also)

Klimbim1, 2017-07-05:
Aaaaand again; without any indication. I did not change anything on the network nor on something else!!!

```
Something caused a factory reset last time. Here's what I know:

I could not authenticate with the access point. This could be a bad password, or an unsupported network type.
```

Phrohdoh, 2017-08-03:
Is this still an issue?

ConnorRigby, 2017-08-03:
It does still seem to be affecting people with distant, or old wifi access points.

 Farmbot does support a wired Ethernet connection also by the way.

ConnorRigby, 2017-08-18:
@Klimbim1 
I could make a setting where Farmbot won't factory reset when it detects that wifi has disconnected. This will allow your bot to reboot properly, and not randomly reboot into factory reset mode, 
BUT 
Farmbot is not expected to operate without internet. If you're bot has no internet you will not be able to control it from the web app, and more importantly it will not be able to sync up with the web app. 
meaning you may have created sequences, but they will not be available to the bot.

the problem here is ultimately in your environment there is something interfering with your Farmbot's wifi chipset. Farmbot's behaviour is unexpected when not connected to the web.

m3talsmith, 2017-08-25:
@ConnorRigby my expectation would be that Farmbot when disconnected would be:

1. Attempts periodic connection attempts on last known SSID or ethx interface
1. Does not reboot unless told to, has kernel level failure, or has hardware failure
1. Continues operating on last known plot or schematics
1. Resyncs when reconnected to update local cache on plot or schematics

These would be expected just from a reliability stand point. Realizing that as Farmbot grows in popularity that you will have more radio noise level issues to deal with, you should prepare for some sort of hardening and recovery plan like suggested above.

ConnorRigby, 2017-08-25:
The problem with that is if you _actually_ want to change your wifi ssid (for example a new router), you would have to re flash the SD card, so the only escape is to reboot into "configurator/recovery" mode. 

We have internally discussed a solution, it just has not been implemented yet.

m3talsmith, 2017-08-25:
I get wanting to change the wifi (though flashing seems extreme). Glad you guys are discussing it internally. I still hold that the reboot should be a rare thing for a consumer to see. Hope you guys get that figured out.

ConnorRigby, 2017-08-25:
A reboot can only happen if a user requests a reboot, or a failed configuration (to allow for reconfiguration) the problem is right now, not connecting to wifi counts as a failed configuration

Klimbim1, 2017-09-10:
> The problem with that is if you actually want to change your wifi ssid (for example a new router), you would have to re flash the SD card, so the only escape is to reboot into "configurator/recovery" mode.

We have internally discussed a solution, it just has not been implemented yet.

One more input on that: Isn´t there a button on the one of the two boards which could be used for such reset purposes? To reflash the SD card each time you are in need of a "hardware factory" reset (because of the beta state of the software) is quite time consuming; whilst pressing a button on the board would be veeeeery comfortable... what do you think?

ConnorRigby, 2017-09-11:
@Klimbim1 I would love to implement such a thing. I started looking into it a while back. Back when i looked into it there was no software access to the raspberry pi's internal pullup resistors meaning that without using a resistor on the pi's GPIO we could not use its gpio. I think that there has been improvements to the GPIO handling libs, so we could do it now, but we do not currently ship with said button, so it would need to be user supplied unfortunately.

wuchangjin, 2017-09-30:
"Failed to connect to mqtt: :server_not_found"
what can i do???

ConnorRigby, 2017-09-30:
Are you self hosting or using farmbot servers? If you are self hosting, it is likely you have misconfigured either the API or the MQTT broker.

snzip, 2017-12-02:
@ConnorRigby , I encounter same issue that farmbot cannot resurgence. 
It have to reconfigurate it to connect its wifi(formbot-xxxx). 
It's very appreciated to fix this issue or get a new solution to retire it. 
thanks,

ConnorRigby, 2018-01-11:
Should be fixed in 6.0.1!

closed 2018-01-11 by ConnorRigby

PeteEmbleton, 2018-05-16:
I am running 6.0.4  with a farmbot that is supplied by a solar panel and small battery.
it keeps disconnecting and when i go to check it out i find that it has lost its config so i need to reset the wifi and the login.
reading the conversation above it sounds like sort of the same thing but you say this was "fixed" in 6.0.1.
is this is the same thing or am i looking for something else?
thanks

ConnorRigby, 2018-05-16:
When you connect to configurator, the first page will have a screenshot of what happened. Feel free to share it if you think there is a bug. Usually the issue is that Farmbot lost internet connection, lost wifi connection, etc.

PeteEmbleton, 2018-05-16:
> We have a setting called "disable factory reset" that you can toggle which will handle your issue i believe.
> Unfortunately we have more use cases for resetting in this situation at the moment. We have a feature planned that should allow us to not reset as the default case in the future.

Awesome, thanks very much

PeteEmbleton, 2018-05-17:
Hi there Conner

I have toggled the setting to disable the factory reset as suggested, hopefully that resolve the issue.

Here is the log from the login screen, I hope it means more to you then it does to me :)

Pete

 

FARMBOT UNEXPECTEDLY SHUT DOWN

Last shutdown reason: econnrefused

 

environment: prod

 

source_ref: 4220540eff07b6497bbd23c9305d7ebf463cfac2

 

target: rpi3

 

Stacktrace:

 

[             {Farmbot.Bootstrap.Supervisor, :actual_init, 3, [file: 'lib/farmbot/bootstrap/supervisor.ex', line: 147]},

 

{:supervisor, :init, 1, [file: 'supervisor.erl', line: 294]},

 

{:gen_server, :init_it, 2, [file: 'gen_server.erl', line: 365]},

 

{:gen_server, :init_it, 6, [file: 'gen_server.erl', line: 333]},

 

{:proc_lib, :init_p_do_apply, 3, [file: 'proc_lib.erl', line: 247]}]

PeteEmbleton, 2018-05-17:
> econnrefused means the server is denying the bot. Are you using my.farmbot.io?

Yes

PeteEmbleton, 2018-05-21:
Hi Connor,

Are you getting other people with problems?

I have disabled the factory reset (thanks for the tip) and now what I am seeing is intermittent sync’s, maybe 2 -3 times a day at all different hours.

On my wifi connection I don’t see the device registered, although the wifi signal is strong, it is almost as if it has given up trying to connect.

I can make it connect by resetting the power but that seems to be a bit drastic.

 

Thanks

pete

 

 

From: Peter Embleton [mailto:peter@embleton.info] 
Sent: Thursday, 17 May 2018 10:25 AM
To: 'FarmBot/farmbot_os' <reply@reply.github.com>
Subject: RE: [FarmBot/farmbot_os] WIFI connection & reconnection (#320)

 

Yes

PeteEmbleton, 2018-05-23:
I am using your server.
are you seeing anyone else with connection issues?
what i am seeing is the farmbot will connect for a few hours and then disconnect, when i look at the wifi the farmbot is not connected, even though the signal is strong.
when i power it off and back on again it works again for a few hours and then disconnects again