FarmBot/farmbot_os #331: Motor movement fails

Klimbim1, 2017-07-05:
Same shit, different version (FB = 4.0.1).

If i move the bot to a certain coordinate, it fails randomly in a sequence.
With encoders on, all kit material, nothing special.

Look here:
https://forum.farmbot.org/t/v4-0-0-software-update/2152/11

Please solve this, it is soooooo unimaginably annoying that this thing even cannot move without errors!

roryaronson, 2017-07-05:
@Klimbim1 please refrain from swearing on GitHub.

The parameter settings you have chosen all seem reasonable, so I am beginning to think you might have issues with your hardware installation that is causing the movement failures. Please check out this post: https://forum.farmbot.org/t/troubleshooting-failed-movements/2212

You might also try increasing the "Max Retries" number to 100, so that FarmBot tries many times to reach its destination.

Klimbim1, 2017-07-10:
I did everything I could @roryaronson in order to solve the problem.

I might check again the belt tension... but as there is no more specific debug message its quite hard to debug. Why can´t you write the axis in the debug log? From where should I know which axis caused the e-stop? What causes exactly the e-stop? The encoder giving feedback that it stalled for the xx many times that I set the threshhold in the config? As xx max. missed steps? The it tries the yy max. retries time to try that movement again before issueing this very unprecise log message?
I now set max. retries to 100 an apparently it worked, but that is only a statistic thing and I did not try many movements since then...

And swearing... I guess you didn´t hear me swear ;) at least it clearly tells you the state of mind of the user at that moment. Whether this should be edited, I doubt it, but you´re the admin...

ConnorRigby, 2017-07-10:
The e stop happens deep in the firmware with no logging resources
unfortunately. (this is due to the nature of the system. There is just not
enough memory and internal resources of the Arduino to log this kind of
information)

The only thing that will cause a stop is missed steps.

Klimbim1, 2017-07-10:
Hey @ConnorRigby, is there really no way of specifying the plug of the encoder which is issueing the stall message (resp. missed steps)? Thats quite sad.... :(
Is there any other way to record current or the like of the motor at that moment? Thus one could judge about the stall of the motor itself.... is there any more information on when the motor stalls? How is that built mechanically?  Is there a slipping coupling (clutch?) inside the stepper motor?

ConnorRigby, 2017-07-10:
I am by far an expert here but

the encoder is optical, so no physical connection like a clutch that would slip. As far as measuring the current, the electronics can't currently do that, and im not sure what would be needed for that. We may be able to at least determine which axis/motor caused the stall tho. @TimEvWw any thoughts on that one?

Klimbim1, 2017-07-12:
Is there any manuals or specs or something that I could read to know how the encoders exactly work? What does the motor do when driving into a barrier...?

TimEvWw, 2017-07-17:
The motor itself is a basic two phase stepper motor. Magnetic fields are applied in a specific order given by the stepper motor driver make the motor take a step. That part is completely done on the RAMPS board. The arduino only send three signals to the RAMPS board: on/off, direction, step. So the software just gives a command to take a step in the right direction. There is no direct feedback if it works or not. Hence the importance of encoder and good calibration. If the mechanical tension is too big for the magnetic fields in the stator, the rotor stalls. It just doesn't turn undetected.

Can we add something like current measurement. Sure, with enough time and money anything is possible. It would need a newly designed board, new drivers, new software for control or motors with integrated measurement and break all compatibility with current stuff and hackability that is a basic feature, but it's possible.

To come back to the money part, I'm sure that Farmbot as a company will be delighted to get an order request to do this ASAP and make a proposal for you. Otherwise if you're willing to wait a bit longer, we can see what is possible in the newer versions.

But as a cheaper solutions at his moment we use the encoder. The current models give a signal every time the optical sensor in the encoder sees a change. It's explained here.

https://en.wikipedia.org/wiki/Rotary_encoder#Incremental_rotary_encoder

The issue here is that there are a few hundred thousand signals coming from the encoder per second, what is a lot to process for an arduino board, but the latest version should manage. The software will then compare the theoretical position for the motor with the measured position of the encoder. There also is the issue that small grains of sand can already cause a small difference between motor and encoder. The difficulty is in making it sensitive enough to know it bumped into something, but not so sensitive that it stops on a grain of sand or a blade of grass. That all depends on your local situation, calibration of stepper drivers, quality of construction, how level the rail is, ...

There are messages that are generated but you don't see these on the message bar because there are too many messages like that and it would drown out any usable information.

closed 2017-09-10 by Klimbim1