FarmBot/farmbot_os #332: since update to 4.0.1, bot is offline

dcolens, 2017-07-10:
* Device: farmbot (self-made)
  * Farmbot OS version: 4.0.1
  * Arduino Firmware version: 

When upgrading to 4.0.1, the farmbot remains offline, we can see it connects to the AP, but does not connect to my.farmbot.io. The farmbot stays connected to the AP (i.e. it does not return to AP mode). Downgrading to 3.1.6 fixes the issue but gives a warning that we should upgrade on my.farmbot.io. The upgrade was attempted twice with the same result. The upgrade was tested both via the my.farmbot.io site and by burning the flash card (twice for each method).

#### Steps to reproduce:

  1. upgrade from 3.1.6 to 4.0.1

#### Observed Results:

  * BOT is offline reported on my.farmbot.io

#### Expected Results:

  * BOT to come back online after the upgrade

ConnorRigby, 2017-07-10:
Hmm very interesting. :mag: investigating.

dcolens, 2017-07-18:
Some console logs for the record (going back to 3.1.6)
`Farmbot[83]: 1970-01-01 00:00:20.400 [info] DHCPManager.EventHandler(wlan0) udhcpc :bound

Farmbot[83]: 1970-01-01 00:00:20.402 [info] nerves_network_interface received :ifchanged and %{ifname: "wlan0", index: 3, is_broadcast: true, is_lower_up: true, is_multicast: true, is_running: true, is_up: true, mac_address: <<184, 39, 235, 30, 138, 100>>, mac_broadcast: <<255, 255, 255, 255, 255, 255>>, mtu: 1500, operstate: :up, stats: %{collisions: 0, multicast: 0, rx_bytes: 10117, rx_dropped: 37, rx_errors: 0, rx_packets: 39, tx_bytes: 296, tx_dropped: 0, tx_errors: 0, tx_packets: 2}, type: :ethernet}

Farmbot[83]: 1970-01-01 00:00:20.435 [info] >> is waiting for linux and network and what not.

Farmbot[83]: 1970-01-01 00:00:20.461 [info] WiFiManager.EventHandler(wlan0) ifup

Farmbot[83]: 1970-01-01 00:00:21.120 [info] Nerves.NetworkInterface - WiFiManager(wlan0, up) got event :ifup

Farmbot[83]: 1970-01-01 00:00:22.763 [info] Initializing Firmware!

Farmbot[83]: 1970-01-01 00:00:24.867 [info] >> is reading all mcu params.

Farmbot[83]: 1970-01-01 00:00:24.928 [info] >> read all params try 1

Farmbot[83]: 1970-01-01 00:00:25.134 [info] >> Read all params

Farmbot[83]: 1970-01-01 00:00:25.507 [info] >> doing connection test...

Farmbot[83]: 1970-01-01 00:00:26.503 [info] >> connection test complete

Farmbot[83]: 1970-01-01 00:00:26.504 [info] >> starting ntp client.

Farmbot[83]: 1970-01-01 00:00:29.144 [info] >> is getting time from NTP.

Farmbot[83]: 1970-01-01 00:00:29.145 [info] >> trying to set time (try 0)
`

ConnorRigby, 2017-07-18:
Im not able to reproduce this one. i can still update from 3.1.6 -> 4.0.0 (manually) and to 4.0.1 (via the ota button on the web app and manually) and i am able to manually flash the 4.0.1 image. 

do you have console logs from bootup on 4.0.0 or 4.0.1?

dcolens, 2017-07-19:
I do but it contains SSID, password, IP addresses so I’d need to clean it first before sharing

Did

dcolens, 2017-07-19:
here's the boot log https://pastebin.com/LpydxyE4

ConnorRigby, 2017-07-31:
@dcolens sorry for the long wait.
Are you still unable to connect on 4.0.1?

dcolens, 2017-07-31:
we gave up and reverted to 3.1.6 is there any reason why 4.0.1 would work now ? As far as we could test there is absolutely no way to get 4.x.y running.

ConnorRigby, 2017-07-31:
Hm. very strange, I'm looking thru your pastebin again. It all looks normal to me until the log is cut. If that is where the TTY _just stopped_, it looks to me like it lost power at bootup

ConnorRigby, 2017-07-31:
Actually just noticed something. It looks like, its having a hard time getting time from NTP for some reason. 
any your bot fully boots up on `3.1.6`? NTP hasn't changed in about 9000 years. Weird that that could fail at all.

ConnorRigby, 2017-07-31:
Do you have a slow or intermittent Wireless connection?

dcolens, 2017-07-31:
yes, bot runs fine on 3.1.6 always boots ok, with 4.0.x it just never came online. AP is in view of the farmbot and is dedicated to the farmbot, we also have a camera on the same network and it has no connectivity issues so we think networking is ok.
How is NTP configured, is it a hard-coded ntp server or does it use the info from DHCP ?

ConnorRigby, 2017-08-14:
@dcolens Have you tried 5.0.x yet?

ConnorRigby, 2017-08-18:
Closing due to inactivity. @dcolens if this is still an issue, please reopen.

closed 2017-08-18 by ConnorRigby

dcolens, 2017-09-21:
Sorry for getting back to you so late, I was able to test 5.0.2 today and the bot is online ! I confirm the issue is fixed in 5.0.2 :)