FarmBot/farmbot_os #333: Homing error

Klimbim1, 2017-07-13:
Farmbot 4.0.1. 

While homing, the z axis was at the hard stop and the field in the controls screen showed decreasing z values. Although the axis couldn´t move anymore because it was on the stop.

During one sequence the z axis is set to 0.

ConnorRigby, 2017-07-31:
Sorry, I forgot to respond to this when I first saw the email.

When you are homing are you trying to use encoders or end stops?
I'm not 100 percent sure, but i believe you can only use one or the other.

Klimbim1, 2017-09-10:
I was trying to use encoders; at the moment (5.0.1) this did not appear anymore, however I did not spend much time with working in 5.0.1.:

proposal to close this thread

closed 2017-09-11 by ConnorRigby