FarmBot/farmbot_os #346: [question] Considering ROS

archenroot, 2017-08-19:
Hi,

have you considered Robot Operating System as base for this amazing project?

ConnorRigby, 2017-08-19:
Yes. Unfortunately they are not quite suitable for a consumer friendly project.
* There is no deploy pipeline for developers.
    * This means there is no simple way of creating a lean stable image to be flashed to an SD card, and doing so becomes a manual task for the developer
* It is developed in python.
    * Python is a scripting language, and I've found not very stable or suitable for production systems. This is mostly personal preference.

closed 2017-08-19 by ConnorRigby

archenroot, 2017-08-19:
I actually agree on both points :-)

Regarding the pipeline, there is some project if I am not wrong regarding its purpose:
https://github.com/mikepurvis/ros-bundling

Regarding Python, well, I have have seen lots of production grade systems on Python, but I do not prefer this language as well. I am experimenting for some while with OpenHAB (openhab.org), which is Java framework for house automation as this is related I had in mind pure Java based on ROSJava.

But I would understand that Java is not your language of choice...

ConnorRigby, 2017-08-19:
That bundler you linked is close to what is needed, but not the entire package. 
I prefer to do all my work on the host machine, then deploy to the target. 
So I can build, test and debug locally on my 8 core, 64 gigs of RAM machine. The when its ready I can easily do something that will:
Cross compile a Linux environment, cross compile all of my libraries, cross compile my project, then _finally_ bundle to one single file to be distributed to the target device. (in this case the raspberry pi). 
I've only seen one framework that does this entire thing cross platform and that is nerves. There is other options. You suggested Java. Java actually can be deployed super easily because it is a virtual machine. But java misses some things also
From the farmbot os runtime, we require networking configuration utilities. As far as I know, there aren't any Java libs for configuration of network. (besides maybe Android?) Plus I just don't really like java. C/Cpp projects have a similar argument but again you have to build the entire development pipeline yourself, and I prefer a higher level than c/Cpp when it is applicable.

Also our "robot" code is actually wrote in Cpp and executed on an Arduino. It communicates to farmbos os via uart, so any ROS would be a little bit overkill.

Ritesh1991, 2017-10-26:
it is possible to to implement ros on linux for flash image and use it to production device. if you provide documentation of farmbot os development than we can do it. implement whole elixir package on ros. and than bind it with linux ready to use as plug and play