FarmBot/farmbot_os #348: Can't ssh to device on 5.x

rootux, 2017-09-19:
I don't have the option to enable SSH on the configuration screen
I've also tried putting the ssh file on the image and it did not work

rootux, 2017-09-19:
Version 5.0.2 (Latest) This is how my configuration screen looks like. No ssh.

[image]

ConnorRigby, 2017-09-20:
I've updated the [docs](https://github.com/FarmBot/farmbot_os/blob/master/docs/FAQ.md#can-i-ssh-into-the-farmbot). 

Basically FarmbotOS has no SSH because there are no usable Linux Utilities. (IE: `raspi-config`, `apt-get`, `sudo` etc)

closed 2017-09-20 by ConnorRigby