FarmBot/farmbot_os #349: Take Photo Fails On Raspberry pi Camera V2.1 on 5.0.2

rootux, 2017-09-19:
1.Connected the camera.
2.Set the Camera to `Raspberry Pi camera`.
3.Press on `Take photo`
3.Farmbot [take-photo] Problem getting image.

This is a new Farmbot image 5.0.2 (But it fails on an old image 5.0 after an upgrade)
Camera is version 2.1

gabrielburnworth, 2017-09-19:
Have you tried rebooting?
The error message could also indicate that there's a ribbon cable connection issue.

rootux, 2017-09-19:
I've tried rebooting several times. Will try again and tried to connect the ribbon cable 4 different places. will try again.
Have you tried working with the V2.1? Do you support this camera?

rootux, 2017-09-19:
@gabrielburnworth OK weird now it looks like I get the cable connected right and I'm getting
Orange mark with - `Image Watcher trying to upload /tmp/images/1505859478.jpg~`

rootux, 2017-09-19:
So it just does not show the image anywhere

[image]

gabrielburnworth, 2017-09-19:
Ah, that is a bug that has been fixed in the next FarmBot OS version, 5.0.3, which should be released this week. It looks like your camera is connected properly now, so after you upgrade it should work fine. Thanks for reporting the issue!

gabrielburnworth, 2017-09-22:
FarmBot OS v5.0.3 has been released.

closed 2017-09-22 by gabrielburnworth