FarmBot/farmbot_os #350: Cannot upload image to Aliyun Cloud

yyzzz6000, 2017-09-20:
Hi,
I am using farm bot in China. Because we cannot access Google Cloud Storage, we modified farm web code and switch to Aliyun online storage API which is very similar to Google's but requires that the key "file" must be the last one in the request body form data. 
However, we monitor the request sent by the farm bot and the key "file" is not the last one. After the investigation, I think it is caused by enum map ordering mechanism.
in file http.ex, line 113
mp = Enum.map(form_data, fn({key, val}) -> if key == "file", do: {"file", file}, else: {key, val} end)
This line will change the original order received from farm web storage auth api and order the keys alphabetically. 

I am considering two workarounds:
1. build an additional api to receive the image upload request from farm bot, change the form data key order and forward the request to Aliyun.
2. modify farm bot code to concat %{"file", file} instead of changing the value in line. 
I am not sure about the impact of the second change, and I appreciate any suggestion.

RickCarlino, 2017-09-20:
@yyzzz6000 Thanks for letting us know about this!

You mention that you cannot access GCS in China. Do you mean that you cannot buy services from GCS as a developer, or do you mean that users in China cannot access GCS at all, even if they just want to view images at `my.farmbot.io`?

As for your original question about making a proxy server or changing the order of form data, I will defer to @ConnorRigby .

snzip, 2017-09-20:
@RickCarlino , NO, it's not proxy issue.   in China, Google cloud is not accessable. So we setup own Farmweb site and use Aliyun OSS to replace google cloud storage . But AliyunOssApi requires file as last item in form data, it impact with          farmOS function 
https://github.com/FarmBot/farmbot_os/blob/master/lib/farmbot/http.ex#113

ConnorRigby, 2017-09-20:
I believe I remember GCS requiring the `file` field being the *first* field. I'm testing out if it matters right now. If it doesn't break compatibility with GCS, I have no problem changing it. Will get back to you once I know for sure

ConnorRigby, 2017-09-20:
OK. I was able to reverse the list without problem. @yyzzz6000 @snzip we will have a release out later this week for you to try out.

PR #351 (2017-09-20): More Multiple Firmware files fixes
* Bundle new FW files
* Allow list of expected fw versions
* fix #350
closed 2017-09-20 by ConnorRigby