FarmBot/farmbot_os #361: scripts/build_system.sh problem

cyriacr, 2017-11-18:
Hi, I tried to go thru scripts/build_system.sh but it failed with the following output. Anyone knows how to get pass these file missiing/permission issues?

`--2017-11-17 23:26:11--  (嘗試第  2 次)  https://github-production-release-asset-2e65be.s3.amazonaws.com/63556949/0e5b4312-0328-11e7-85e6-7ee1f3920996?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20171117%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20171117T151756Z&X-Amz-Expires=300&X-Amz-Signature=a233f5ac6fbd90df3519c3203a3f2a2e1019a9e68adb230f4b677269e1d0beaf&X-Amz-SignedHeaders=host&actor_id=0&response-content-disposition=attachment%3B%20filename%3Dnerves_toolchain_arm_unknown_linux_gnueabihf-0.10.0.linux-x86_64.tar.xz&response-content-type=application%2Foctet-stream
正在連接 github-production-release-asset-2e65be.s3.amazonaws.com (github-production-release-asset-2e65be.s3.amazonaws.com)|52.216.130.227|:443... 連上了。
已送出 HTTP 要求，正在等候回應... 403 Forbidden
2017-11-17 23:26:12 錯誤 403: Forbidden。

--2017-11-17 23:26:12--  http://sources.buildroot.net/nerves_toolchain_arm_unknown_linux_gnueabihf-0.10.0.linux-x86_64.tar.xz
正在查找主機 sources.buildroot.net (sources.buildroot.net)... 176.9.16.109
正在連接 sources.buildroot.net (sources.buildroot.net)|176.9.16.109|:80... 連上了。
已送出 HTTP 要求，正在等候回應... 404 Not Found
2017-11-17 23:26:13 錯誤 404: Not Found。

package/pkg-generic.mk:128: recipe for target '/home/eddie/farmbot/os/nerves/NERVES_SYSTEM_rpi3/build/toolchain-external-custom/.stamp_downloaded' failed
make[1]: *** [/home/eddie/farmbot/os/nerves/NERVES_SYSTEM_rpi3/build/toolchain-external-custom/.stamp_downloaded] Error 1
Makefile:16: recipe for target '_all' failed
make: *** [_all] Error 2

`

ConnorRigby, 2017-11-18:
Your machine seems to be behind a firewall preventing your machine from accessing the sources.buildroot.net. You will need to make sure buildroot.net and the many related domains are accessable.

closed 2017-11-18 by ConnorRigby

cyriacr, 2017-11-18:
Hi,

Thanks for quick reply.

However, I've double checked it's not network problem.

http://sources.buildroot.net/rpi-firmware-b51046a2b2bb69771579a549d157205d9982f858.tar.gz

That file reallly doesn't exist.

Eddie Lien

Union Internet Technology Co., Ltd.
Inspiring International Limited.

ConnorRigby, 2017-11-18:
Ah yeah upstream rpi-firmware deleted that commit a while back. 
If you `git checkout v5.0.6` this should be fixed.