FarmBot/farmbot_os #409: cannot setup 6.0.0

snzip, 2018-01-09:
START
FARMBOT UNEXPECTEDLY SHUT DOWN
Last shutdown reason: Failed to start child: Elixir.Farmbot.Firmware reason: enoent This is likely a bug. Please copy or screenshot this error and send it to the Farmbot developers.

environment: prod

source_ref: e43026a8af6a83878ead46210031987b54df11e6

target: rpi3

Stacktrace:

[	{Farmbot, :start, 2, [file: 'lib/farmbot.ex', line: 22]},

{:application_master, :start_it_old, 4, [file: 'application_master.erl', line: 273]}]

Get ready to configure your FarmBot!

Network
Firmware
Web App Credentials

ConnorRigby, 2018-01-09:
Can you tell me about your arduino? Is it a genuine Arduino or reproduction?

ConnorRigby, 2018-01-09:
Also for the record - `6.0.1` is still a prerelease. 5.0.9 is the most recent production release.

snzip, 2018-01-10:
yes, it's v6.0.1,   1.2. arduino

ConnorRigby, 2018-01-10:
Can you take a photo of your Arduino?

ConnorRigby, 2018-01-11:
Official 6.0.1 is out. Please repoen if you have any other issues

closed 2018-01-11 by ConnorRigby

rdegosse, 2018-01-31:
Hi, i have the same issue with 6.0.1 / 6.1.2 and my arduino is not genuine.
see: https://forum.farmbot.org/t/dont-config-6-0-1/3117
@ConnorRigby : do you think reproduction arduino can cause this issue?
thanks

rdegosse, 2018-02-02:
issue solved with a new arduino, not genuine but not low cost!