FarmBot/farmbot_os #411: Arduino Firmware Not getting uploaded when starting Farmbot OS on RPi3

dwilli37, 2018-01-10:
We are currently running Farmbot OS 5.0.9 and got all the way through setup and then needed to switch Wifi Routers. So we reloaded the OS on the SD card and waited for Farmbot-XXXX to show up and it did not. Reading through some of the forms it looked like if the Arduino did not initialize correctly then the boot would fail. So I put Blink on the ardunino to see if the Farmbot code was getting loaded and Blink never went away. Is something we should be doing different to get the code onto the arduino?

ConnorRigby, 2018-01-10:
During configuration, FarmbotOS uploads Arduino Firmware during configuration. I recommend you try flashing 5.0.6 (https://github.com/FarmBot/farmbot_os/releases/tag/v5.0.6) then upgrade from the app.

dwilli37, 2018-01-10:
So we put 6.0.1 on and it everything is not working so far.

ConnorRigby, 2018-01-10:
`6.0.1` is still prerelease. It is not ready yet. Please use [5.0.6](https://github.com/FarmBot/farmbot_os/releases/tag/v5.0.6)

ConnorRigby, 2018-01-11:
6.0.1 is out and should fix this if it were to happen to anyone else. Please reopen if you  find any more issues.

closed 2018-01-11 by ConnorRigby