FarmBot/farmbot_os #451: Error Setting Up Farmbot

eamonpenland, 2018-02-20:
Hi All,

I'm having an issue setting up my farmbot.

any ideas?

[image]

gabrielburnworth, 2018-02-20:
The error means that the firmware on your Arduino isn't responding or may be missing. Did you have the Arduino plugged in and connected to the Raspberry Pi during configuration?

eamonpenland, 2018-02-20:
yup i did

gabrielburnworth, 2018-02-20:
It's possible the Arduino isn't working properly. Some troubleshooting options:

If you have access to another Arduino, you can try using a different Arduino during configuration.

If you can plug the Arduino into a computer running the Arduino IDE, you can try opening a serial connection (Serial Monitor) to the Arduino or try uploading a new firmware to the Arduino.

gabrielburnworth, 2018-02-21:
Is the Farmduino power LED on? Do you see two other flashing LEDs during the firmware step of configuration?

eamonpenland, 2018-03-18:
Thanks for your quick response! Sorry it's taken so long to get back to you. The LEDs on the rpi flash during the firmware step of the configuration and the farmduinos power is on.

eamonpenland, 2018-05-16:
i have not. Will try again this weekend. Thanks!

eamonpenland, 2018-05-19:
So i tried the latest release and still having issues 
[image]

eamonpenland, 2018-05-21:
What can I do to help? I have some basic nerves experience and work with elixir regularly

ConnorRigby, 2018-05-21:
You can try booting up with no arduino plugged in at all. This should allow you to at least get up and running. If you have an FTDI cable, there will be logs on the console that are a little bit more descriptive of what happened.

ConnorRigby, 2018-07-03:
@eamonpenland did you ever get your device booted?

ConnorRigby, 2018-07-27:
Closing due to inactivity.

closed 2018-07-27 by ConnorRigby