FarmBot/farmbot_os #452: Error/bug

kezzab25, 2018-02-23:
[image]

Any ideas why this is happening?

gabrielburnworth, 2018-02-23:
We've seen this occur on Arduinos with power or connection issues. The Arduino starts a connection, then immediately closes it. You can try connecting your Arduino to your computer to verify the issue, or try configuring with a different Arduino. Which type of Arduino are you using?

kezzab25, 2018-02-23:
I have tried 2 different types of arduinos, both with external power
supplies. They both work fine on a computer.

gabrielburnworth, 2018-02-23:
Which two types have you tried?

kezzab25, 2018-02-23:
rcube and a generic. I also cant upload via the Arduino IDE as there are multiple errors

gabrielburnworth, 2018-02-23:
Can you provide the information the Arduino IDE lists under `Tools > Port` for each of those boards? `VID` and `PID` from `Get Board Info` would also be helpful.

kezzab25, 2018-02-23:
Generic - 
BN: Unknown board
VID: 1A86
PID: 7523

Rcube - 
BN: Unknown board
VID: 0403
PID: 6001

gabrielburnworth, 2018-02-23:
Thank you for that info. Do those both use the `ATmega2560 (Mega 2560)` processor?

kezzab25, 2018-02-23:
Yes they do. Is it a problem that the Pi is seeing them as ttyUSB0

kezzab25, 2018-02-24:
I managed to get it working. I updated the bootloader and it all seems to be ok now

kezzab25, 2018-02-24:
I updated the bootlader on the mega and it worked until I did an update on the web-os. Now i get the error, 
[image]

kezzab25, 2018-02-24:
I am not sure what is wrong now. I reflashed the Pi back to the factory image and now it has gone back to its original issue.

kezzab25, 2018-02-26:
Is someone able to help? I have tried other boards, extra power supplies and still it wont work. What am I doing wrong?

ConnorRigby, 2018-02-26:
You may have a faulty pi, but since you said your PC had trouble communicating with the Arduino, it could be the cable, or some unknown factor. Then try configuring Farmbot with just the Arduino, (no ramps shield).

For more information it might be helpful to see a picture of your setup

gabrielburnworth, 2018-02-26:
Also, did any of the Arduinos you tried have a recognized board name? (instead of `BN: Unknown board`)

kezzab25, 2018-02-26:
My computer is working fine talking to the Arduino. I found out one was a
1280 and that's why 1 didn't work but I have 2 more. One worked fine until
the pi did an update and now it won't work again. The pi is fine and can
see the Arduino. I plugged a video cable into the pi to see what was
happening

kezzab25, 2018-02-26:
They are all unknown as I don't have any genuine 2560 boards.

ConnorRigby, 2018-02-26:
The update isn't what caused the issue, the issue was the reboot. Something is causing the USB to Serial converter chip on the clone Arduino reset. Farmbot OS is detecting `/dev/ttyUSB0` then it is resetting while the driver is initializing, and probably showing up as `/dev/ttyUSB1`. 

A number of things can cause it, most commonly bad power.

ConnorRigby, 2018-02-26:
Also, I don't think you will see any information on video. I think that all gets disabled.

kezzab25, 2018-02-26:
I am seeing ttyUSB0 on the monitor but no mention of UBS1.  The power supply is ok as it can handle 10 Amps.

ConnorRigby, 2018-02-26:
Can I get a picture of both your setup and the output on the monitor if at all possible?

kezzab25, 2018-02-26:
[image]

kezzab25, 2018-02-26:
I'm also trying to get it working on a MKS Base v1.5 but I am still getting the same issue. It is a ftdi USB interface

kezzab25, 2018-02-26:
I just installed version 5.0.9 and for the first time it uploaded the firmware to the arduino, it has never done that before and it is also working fine now. How long can i tell it not to update before it stops working?

ConnorRigby, 2018-02-26:
The same issue will happen upon reboot unfortunately regardless of updates. 

We usually support old releases for a few months. I will try to figure out if it is a Linux driver issue in the meantime

kezzab25, 2018-02-26:
That would be good if you could find out what's happening. It would be nice to be able to use the MKS board as its an all in one and not a shield.

ConnorRigby, 2018-02-26:
Can we get a picture of the arduino board you are using? The log you captured above is displaying the usage of a CH341 chip, but the board you are talking about _should_ come with a genuine USB Driver Chip FT232RL.

kezzab25, 2018-02-26:
I have changed boards since that screenshot. And yes, the board comes with a FT232 chip. I am on a mobile device and not sure how to attach an image

ConnorRigby, 2018-02-26:
So you are saying that the `FT232` board does *not* work with `~> 6.x` releases? 

As for an image you can use external upload services such as [imgur](https://imgur.com/upload)

Id like to see a screen capture of the monitor on `6.0.1` _and_ a picture of the board that is not working if possible.

kezzab25, 2018-02-27:
That is correct. the FT232 board is not working with v6.x. 
I have attached images of the setup and what is on the monitor
[image]
[image]

kezzab25, 2018-02-27:
Have you had any luck? It would be nice to get it working with the latest FW

ConnorRigby, 2018-02-27:
Okay so this turned out to be partially a bug on the FBOS side. It will be fixed in the next release, but in the meantime here is what you can do:

* Flash your Arduino manually with the latest firmware. 
   * If you need help with this i can walk you through it, there are a number of ways you can accomplish this.
* Flash your device with 6.0.1
* Boot and configure Farmbot **without** your Arduino device plugged in.
* Upgrade using the web app.
* Wait until Farmbot is fully booted up, and maybe do a `sync` to make sure everything is up and running. 
* Plug your Arduino
* Wait a few seconds, and you should be able to use it.

ConnorRigby, 2018-02-27:
For viewers at home [here](https://github.com/FarmBot/farmbot_os/blob/v6.2.0/lib/farmbot/firmware/uart_handler/auto_detector.ex#L36) is the line that is currently messed up.

kezzab25, 2018-02-27:
Thank you. I will try that tonight when I get home and let you know how I go.

ConnorRigby, 2018-02-27:
Depending on your timezone, we will probably have a proper release out by "tonight"

kezzab25, 2018-02-28:
It worked to begin with, right up until I plugged the Arduino board back in. I took a screen shot of the log on my.farmbot and the web gui on the Rpi.
[image]
[image]

kezzab25, 2018-02-28:
It appears to be working. I had issues but I forgot to change the board type in Boards.h.

ConnorRigby, 2018-03-05:
Sorry for the delay. This should be fixed in the [latest release](https://github.com/FarmBot/farmbot_os/releases/tag/v6.2.1)

You will have to flash 6.0.1 first, without your Arduino plugged in, upgrade until there are no updates available, then you can plug your Arduino in and it should work properly and persist reboots. I will have more elaborate instructions in the FAQ if you are having trouble.

ConnorRigby, 2018-03-05:
Here is the relevant [FAQ](https://github.com/FarmBot/farmbot_os/blob/staging/docs/FAQ.md#non-genuine-arduino-boards-not-detected) entry.

closed 2018-03-05 by ConnorRigby