FarmBot/farmbot_os #509: Sequence looses body after PATCH

etcipnja, 2018-05-05:
After creation: [{u'kind': u'move_absolute', u'args': {u'speed': 4, u'location': {u'kind': u'coordinate', u'args': {u'y': 2, u'x': 1, u'z': 3}}, u'offset': {u'kind': u'coordinate', u'args': {u'y': 0, u'x': 0, u'z': 0}}}}]
After PATCH: []
'''
import os
import requests
import json

headers = {'Authorization': 'Bearer ' + os.environ['API_TOKEN'],
           'content-type': "application/json"}

sequence={
  "name": "Scare Birds",
  "body": [
    {
      "kind": "move_absolute",
      "args": {
        "location": {
          "kind": "coordinate",
          "args": {
            "x": 1,
            "y": 2,
            "z": 3
          }
        },
        "offset": {
          "kind": "coordinate",
          "args": {
            "x": 0,
            "y": 0,
            "z": 0
          }
        },
        "speed": 4
      }
    }
  ]
}

try:

    #create sequence
    response = requests.post('https://my.farmbot.io/api/sequences', headers=headers, data=json.dumps(sequence))
    response.raise_for_status()
    print(response.json()['body'])
    #update sequence name
    response = requests.patch('https://my.farmbot.io/api/sequences/{}'.format(response.json()['id']), headers=headers, data=json.dumps({'name':'dude, where is my body'}))
    response.raise_for_status()
    #body is gone!
    print(response.json()['body'])

except requests.exceptions.HTTPError as error:
    print('HTTP error {} {} '.format(error.response.status_code, error.response.text))
'''

RickCarlino, 2018-05-07:
@etcipnja Thank you for your detailed error report. I will investigate the issue shortly.

RickCarlino, 2018-05-07:
@etcipnja The issue is fixed here: https://github.com/FarmBot/Farmbot-Web-App/pull/836

Unfortunately, the example you have shown above will not work with the API in its current form. The proposed change to the app (above) fixes the issue by providing an error message to the API consumer.

Making `body` and `args` mandatory on update is due to a variety of reasons, mostly server performance (sequence updates are the slowest operation on the API, currently) and supporting devices that are on old versions of FarmBot OS.

In the future, if you wish to perform an update a sequence, you will need to download the entire sequence (`GET /api/sequences/123`) before proceeding.

closed 2018-05-07 by RickCarlino