FarmBot/farmbot_os #536: Save farmbot events into a database for further analysis

derletztename, 2018-05-13:
Save farmbot events like how often and how long is one plant being watered, how much water was consumed by it over what time and how much time is spend on weeding in what area for better development of the algorithms of farming.
Have a continously growing table that can also be read by none coders via the web app. 
Maybe even implement analysis tools for optimisation later or feed it into some machine learning algorithm that learns which plants needs which kind of treatment in which part of the world.
Nevertheless the database would be the first step in this direction, I guess.

derletztename, 2018-05-15:
Hi RickCarlino,

that's awesome to hear :)
Can I already use this in my farmbot?
It could be offline in the beginning, I don't mind. But it really would help to already store the information somehow as I need to start seeding soon. Natur is relentless ;)

RickCarlino, 2018-05-15:
@derletztename I haven't tried in a while, but it might be possible to store the readings via email:
One idea:
1. Perform a "read pin" on the desired pin.
2. Send the data to email via the `send_message` block. For example, to email the status of `pin16` you could send a message such as `"Pin 16 value is {{ pin16}}"`

You could also write a custom farmware, if you are familiar with the Python programming language.