FarmBot/farmbot_os #600: celery script emergency_unlock fails..

muenchris, 2018-07-25:
I have my local access software ready to do almost all tasks I need to do on a regular base even if the cloud (my.farmbot.io) is down.

The only functions I cannot get to work right are "emergency_unlock/lock".
How can this be called from celery script? The error I get is:

FunctionClauseError at POST /api/v1/celery_script 
no function clause matching in Farmbot.CeleryScript.AST.decode/1
Called with 1 arguments
%{"kind" => "emergency_unlock"}

Thanks
Chris

ConnorRigby, 2018-07-25:
What does the JSON look like? it looks like you are missing the `"args"` field.

try: `{"kind": "emergency_unlock", "args": {}}`

muenchris, 2018-07-25:
args is null (not set) since its not used. Is args always required even if its not used?

muenchris, 2018-07-25:
I am using Newtonsoft to serialize my classes ..."args":{} is not something C# can produce in any form with Newtonsoft JSON serialization. I need to find a way to tell Newtonsoft to produce that output....

ConnorRigby, 2018-07-25:
Yes it is. See [here](https://developer.farm.bot/docs/celery-script) for a description of the protocol.

Closing for now. Feel free to re open if adding args does not help.

closed 2018-07-25 by ConnorRigby