FarmBot/farmbot_os #616: Sequences with wait(60000) can't be executed from the API

etcipnja, 2018-08-10:
# Expected Behavior
Sequence with wait(60000) command (or longer) can be executed from the API

# Actual Behavior
```
node = {'kind': 'execute', 'args': {'sequence_id': sequence['id']}}
response = requests.post(self.farmware_url + 'api/v1/celery_script', data=json.dumps(node),
                        headers=self.headers)
```
second line generates exception 
```
('Connection aborted.', BadStatusLine("''",))
```
# Steps to Reproduce
- create a sequence open watering, wait 60+ sec, stop watering
- run the script above - you get exception
- change the wait duration to 59s, run script - no exception  

note: any sequence can be executed from the web page without any problems. 

Version 6.4.8

ConnorRigby, 2018-08-10:
This appears to be a bug in python?
* https://bugs.python.org/issue8450
* https://bugs.python.org/issue8450

I'm updating underlying Linux system deps today. I'm not sure if it has the fix of not, but i'll update this issue when it's out and test it out.

ConnorRigby, 2018-08-10:
Turns out there is no fix for this in python2 from what i can tell. I can probably increase the http keep-alive header, but i'm not sure that is a solution, more of a hack/patch. 

This is sort of a problem with HTTP itself, and us overloading its intended functionality, although i'm not sure of many other solutions that are simple to implement.

ConnorRigby, 2018-08-10:
Also for what it's worth, anything celeryscript node that takes longer than 60 seconds will be broken. (this includes long movements, compound short movements etc)

etcipnja, 2018-08-10:
I propose to increase timeout as a short term remedy. This feature worked before and got broken few weeks ago. Some people were affected as you can see

ConnorRigby, 2018-08-13:
Fixed in #620

closed 2018-08-13 by ConnorRigby