FarmBot/farmbot_os #640: Please consider vm.args : +C multi_time_warp

jsimmonds2, 2018-09-21:
Hi !
Been having a head-scratching time trying to keep a little Process.send_after (monotonic {:abs, true}) in step with the OS time (which is now carefully NTP synchronized) . .  **Until** I noticed that the VM is started with the default `:no_time_warp` mode.

The ERTS guide to Time and Time Correction strongly encourages adopting the new API.  There wouldn't be old code still using erlang:now/0 _would there_ ?

ConnorRigby, 2018-09-21:
This sounds like a great idea. I always use millisecond offsets in Farmbot code, but sub deps will need to be checked.

ConnorRigby, 2018-10-08:
I'm testing this out today. It seems to be working alright. I'm going to do some QA and report back

ConnorRigby, 2018-12-03:
This is staged to be released. :+1:  Thanks @jsimmonds2

closed 2018-12-03 by ConnorRigby