FarmBot/farmbot_os #644: Farmbot seems to stall on boot?

scottwegner, 2018-10-29:
I'm really struggling here desperate for help... 

I got a farmbot for our school / students to work on.  I've got it together... but I have never been able to keep it connected reliably.

I think it was, at various time a variety of things.  I thought I had solved the issue when I put a hole in my district firewall so big a mac truck could drive through it - but then mysteriously the dreaded "Disconnected" message came on again in the web portal.

I've re-flashed it several times altering one setting or another.

This last time I finally figured it was something to do with maybe the wireless was not working so I re-installed the OS again and selected Wired Ethernet - with a static IP.  I still could not PING that ip... so I hooked the unit up to a monitor just to get some sort of visibility - and it seems to be hung up on boot at a spot that says: 

IPV6 ADDERCONF (NETDEV_CHANGE:) WLAN0: link becomes ready

after that NOTHING?

anyone have any advice?  

One other note - I do a fair amount of work with networking / Open Source / Raspberry PI / etc.  I am the network admin here...  I'm kind of stumped.  

Thanks!
Scott

ConnorRigby, 2018-10-29:
So It probably only looks like FarmbotOS is hanging because the info on the HDMI port is only the kernel log. (IE the output of `dmesg -w`). The _real_ logs are on the serial console. To view them you will need [something like this](https://www.digikey.com/product-detail/en/ftdi/TTL-232R-RPI/768-1204-ND)

I've not tested static ip config very much, but i don't see why it _wouldn't_ work. That being said i do use DHCP wired config most often and i never have any issues. 

1) What version of FarmBot OS are you using?
2) Can you try using DHCP on a wired connection if that is a possibility?
3) When FarmbotOS "Disconnect"s does it return to configuration mode? 
   * Does a reboot bring FarmBot back online?

scottwegner, 2018-10-29:
Kernel log makes sense... Any way to get to the console without the FTDI cable?  I would feel so much more comfortable if I could just get into it to look around... it feels so much like a mysterious black box.

I just tried DHCP wired with same result.  I have another brand new Rpi3 - same result.  I've been downloading other images 6.4.8, 6.4.10, 6.4.9, 6.4.11 - pretty much same with all of them.

I think the latest one I tried was 6.4.10.

On your question 3 - I noticed that on this last trial it did keep going back into config mode when DHCP wired didn't work.  I think that image was 6.4.10.

Reboots do not seem to help.

I noticed that consistently when it gets to the step where it flashes the Farmduino - the sequence is kind of "odd":

- I hear subtle clicking as farmduino board resets... some LEDs go off...
- it starts flashing - progress bar moves - LEDs flash fast
- then before the progress bar gets to the end (about 55 - 60 percent across) it quits
- board subtle click - leds go back to normal.  This last few times the Z axis stepper driver is engaged green solid on.
- and the setup wizard continues on and asks for my online web credentials.

I doubt that could be the problem... but is seems odd that the flash appears to bail out early. 

Scott

ConnorRigby, 2018-10-29:
> Kernel log makes sense... Any way to get to the console without the FTDI cable? I would feel so much more comfortable if I could just get into it to look around... it feels so much like a mysterious black box.

For security reasons, most ways of "getting in" to FarmbotOS are closed by default. 
You can follow [this](https://developer.farm.bot/docs/farmbot-os#section-setting-up-ssh) guide to get
an SSH console open, or you can purchase an FTDI cable to see logs.

> I noticed that consistently when it gets to the step where it flashes the Farmduino - the sequence is kind of "odd":

Yeah that progress bar doesn't really mean anything. it's actually just a timer in the background.
The leds flashing quickly means the firmware is being flashed, which is ok. If it moved to the next page, you are likely in good shape.

Next time your device resets to configurator, can you click "Configure your FarmBot" like 10 times, and click the "DOWNLOAD ALL LOGS" button, then when the "Log Viewer" page loads, click "DOWNLOAD ALL LOGS" button again. 

this will create a file with a `.sqlite` extension. 
Can you email that to me? (connor<atsymbol>farmbot.io) 
(this _may_ have your network information embedded in it if you used wifi, since your problem seems to happen on wired connection as well, i suggest using this)

scottwegner, 2018-10-30:
I think I can troubleshoot it myself... but other than SSH and a serial cable... how do I re-enable the serial console.  I've been booting up another Pi and editing the cmdline.txt of the farmbot SD card that way... but I have not found the solution.  Most places have a "use raspi-config" solution ... which is not an option in this case.   So two things:

- what was done to disable login in the first place so I can reverse it?
- what is the default command line login?  (that would save me having to hack around in the passwd file)  :-)

Scott

ConnorRigby, 2018-10-30:
The setting for the console is set at compile time, and to change it requires recompiling FarmbotOs.

There is no login (ie getty) as there is only one user (root). This is similar to openElec or ddwrt, or openWrt for example.

There are also almost no regular Linux utilities save a few very important ones. By extension of this there is no shell (ie bash), no `apt-get`, no raspi-config etc.

You are welcome to recompile Farmbot OS with the console on HDMI or I can build you a firmware with this set-up, however if you are going to be working woth FarmbotOS internals, I highly suggest using an FTDI adapter. That is industry standard forany years and is the most convenient way of getting data to/from FBOS without doing things like manually copying logs, taking pictures etc

scottwegner, 2018-10-30:
I ordered an FTDI cable just now.

I don't intend to spend any time working on internals ... but I do know that I've got two classrooms of kids wondering why Mr. Wegner (me) can't seem to keep the farmbot connected to the my.farmbot.io site.  So out of desperation I'm looking for any way to get visibility into what is going wrong.

I know that it must sometimes connect, because I've been able to move it and do other things... but then mysteriously - it all ends.  

So if you were me, and you wanted to do some very basic network level troubleshooting - where would you start?

So sorry to be "that guy" - I don't mean to be a pest.  Just trying to troubleshoot a black box.  :-)

Scott

scottwegner, 2018-10-30:
is there any console indication what DHCP address it grabbed?

ConnorRigby, 2018-10-30:
> So sorry to be "that guy" - I don't mean to be a pest. Just trying to troubleshoot a black box. :-)

No need to apologize. Farmbot's stability in enterprise settings is something we know can be improved, so troubleshooting these issues is important.   

The fact that it connects at all leads me to believe that DHCP isn't exactly the issue here, i suspect it is FarmBot's message broker connection. Some school networks have issues with connections that stay open for long amounts of time. (IE Farmbot being connected literally all the time).

The steps I laid out [here](https://github.com/FarmBot/farmbot_os/issues/644#issuecomment-434100156) would be really helpful in seeing exactly why FarmBot is disconnecting. 

Is there anything special about your network?
* No proxy?
* Opened the ports described [here?](https://developer.farm.bot/docs/for-it-security-professionals)
   * I assume that since you get connection sometimes, this isn't the issue either
* local DNS server?

The most common issue i've seen when using a wired connection on an enterprise network is the long running connection issue i described above and DNS issues.

scottwegner, 2018-10-30:
Cool.  

Here is what I know now...

I booted that Pi with a generic raspian to get the MAC addresses of eth0 and wlan0

I'm booting  6.4.11  just now to troubleshoot. 

Ran through the config wizard on wifi.  Used DHCP / wireless.  

grep-ing the dhcpd.leases on my dhcp server showed the IP - I pinged it successfully.  It was connected to the local wifi.   Checked the my.farmbot.io site on a whim - no connection.

Just to be SURE that the IP I was pinging was the ACTUAL ip of the farmbot pi... I pulled the power on it and sure enough the pings died.

I plugged it back in and was surprised to find that after boot I could not ping it again!

I poked around and noticed that he farmbot config wifi was up again - So I connected to it again and it was asking me to start the config over.

Which I did... same settings.... and after reboot - the IP could be ping'd again.

But I've never seen anything like you describe here:

"Next time your device resets to configurator, can you click "Configure your FarmBot" like 10 times, and click the "DOWNLOAD ALL LOGS" button, then when the "Log Viewer" page loads, click "DOWNLOAD ALL LOGS" button again." 

Not sure where to go here.

A different ver? 

Scott

ConnorRigby, 2018-10-30:
When the Farmbot wifi is up, the menu is hidden. You will have to tap "Configure your FarmBot" 10 times then there will be a hidden menu. 

Since FarmBot wasn't able to connect to the web services, it resets because it thinks it's config was invalid. 
It is possible that your device does not have a valid time and therefore is hanging on an SSL connection.

scottwegner, 2018-10-30:
wow - the "configure your farmbot" is kind of hidden... found it after many times of not finding it...

Attached.   

[6.4.11-2EB84A4CA1C33E40157787F51887A34C-logs.zip](https://github.com/FarmBot/farmbot_os/files/2530873/6.4.11-2EB84A4CA1C33E40157787F51887A34C-logs.zip)

ConnorRigby, 2018-10-30:
Okay after having a look through those logs, i don't see anything critically wrong. 
Your device is certainly getting an ip address. It looks to me like your device is unable to sync type with `pool.ntpd.org`. It also looks like your bot is unable to detect your arduino. 

Are you using a Farmbot Farmduino? A regular Ramps board? 
Have you tried configuring FarmBot **without** a arduino/farmduino plugged in at all? If not that is a good way to make sure that is not the problem

scottwegner, 2018-10-30:
Ya - the last couple of boots was just the pi alone on my bench.  No Farmduino.  (mine is v1.3)  I started booting it on the bench just so that I didn't have to run back and forth while troubleshooting.

I figured if I can get to the point where I can get it to reliably connect with my.farmbot.io (mqtt etc) - then I'd put it back in place with the Farmduino 1.3

I'll look at the ntp issue with my firewall and see....

Scott

ConnorRigby, 2018-10-30:
Yeah network configuration is something that really needs a facelift. If your network has an NTP server on it, you can also configure that currently in the "Advanced settings" section of network config.

Glad you got it sorted out!

closed 2018-10-30 by ConnorRigby