FarmBot/farmbot_os #646: connection to web app lost intermittently.

scottwegner, 2018-11-20:
This may be a continuation with an issue I posted a while back... I'm not sure.  I can't seem to get my farmbot to stay connected on my school network.  It seems like I'm caught in an endless loop of booting... configuring... brief connection... some random error.. no connection again... only to find the farmbot wifi active again... and re-configuring again... etc etc...  I've attached some screenshots from after a crash - what the configure page shows me.   Any help would be much appreciated! 

[image]
[image]
[image]

ConnorRigby, 2018-11-21:
Huh it looks like a slow connection, is breaking things. I've seen this before in a couple different incarnations. Do you have `auto_sync` enabled by chance? If so you will likely have better luck disabling it.

ConnorRigby, 2018-11-28:
Closing due to inactivity.

closed 2018-11-28 by ConnorRigby